MarlinFirmware/Marlin #21428: [BUG] commit a8ea6e6 causes compile error

lightmaster, 2021-03-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Commit a8ea6e6 causes a compile error. Reverting this commit allows `bugfix-2.0.x` to compile successfully.

### Version of Marlin Firmware

bugfix-2.0.x as of 2021-03-23

### Bug Description

Commit a8ea6e6 causes a compile error. Reverting this commit allows `bugfix-2.0.x` to compile successfully.

```C
Compiling .pio\build\LPC1769\src\src\lcd\menu\menu_info.cpp.o
In file included from Marlin\src\lcd\menu\menu_advanced.cpp:31:
Marlin\src\lcd\menu\menu_advanced.cpp: In function 'void menu_advanced_temperature()':
Marlin\src\lcd\menu\menu_advanced.cpp:264:32: error: cannot convert 'float*' to 'TMenuEditItem<MenuEditItemInfo_int3>::type_t*' {aka 'short int*'}
  264 |       EDIT_ITEM(int3, MSG_MIN, &planner.autotemp_min, 0, thermalManager.hotend_max_target(0));
Marlin\src\lcd\menu\menu_item.h:261:39: note: in definition of macro '_MENU_INNER_P'
  261 |     MenuItem_##TYPE::action(plabel, ##V);                      \
      |                                       ^
Marlin\src\lcd\menu\menu_item.h:356:57: note: in expansion of macro '_MENU_ITEM_P'
  356 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:395:58: note: in expansion of macro 'MENU_ITEM_P'
  395 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:396:58: note: in expansion of macro 'EDIT_ITEM_P'
  396 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_advanced.cpp:264:7: note: in expansion of macro 'EDIT_ITEM'
  264 |       EDIT_ITEM(int3, MSG_MIN, &planner.autotemp_min, 0, thermalManager.hotend_max_target(0));
      |       ^~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:95:22: note:   initializing argument 2 of 'static void TMenuEditItem<NAME>::action(const char*, TMenuEditItem<NAME>::type_t*, TMenuEditItem<NAME>::type_t, TMenuEditItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuEditItemInfo_int3; TMenuEditItem<NAME>::type_t = short int; screenFunc_t = void (*)()]'
   95 |       type_t * const ptr,                   // Value pointer
      |       ~~~~~~~~~~~~~~~^~~
Marlin\src\lcd\menu\menu_item.h:266:59: error: no matching function for call to 'TMenuEditItem<MenuEditItemInfo_int3>::draw(bool, int8_t&, const char* const&, float*, int, uint16_t)'  
  266 |       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);   \
      |                                                           ^
Marlin\src\lcd\menu\menu_item.h:272:5: note: in expansion of macro '_MENU_INNER_P'
  272 |     _MENU_INNER_P(TYPE, ##V);         \
      |     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:356:57: note: in expansion of macro '_MENU_ITEM_P'
  356 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:395:58: note: in expansion of macro 'MENU_ITEM_P'
  395 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:396:58: note: in expansion of macro 'EDIT_ITEM_P'
  396 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_advanced.cpp:264:7: note: in expansion of macro 'EDIT_ITEM'
  264 |       EDIT_ITEM(int3, MSG_MIN, &planner.autotemp_min, 0, thermalManager.hotend_max_target(0));
      |       ^~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:85:30: note: candidate: 'static void TMenuEditItem<NAME>::draw(bool, uint8_t, const char*, TMenuEditItem<NAME>::type_t*, ...) [with NAME = MenuEditItemInfo_int3; uint8_t = unsigned char; TMenuEditItem<NAME>::type_t = short int]'
   85 |     FORCE_INLINE static void draw(const bool sel, const uint8_t row, PGM_P const pstr, type_t * const data, ...) {
      |                              ^~~~
Marlin\src\lcd\menu\menu_item.h:85:103: note:   no known conversion for argument 4 from 'float*' to 'TMenuEditItem<MenuEditItemInfo_int3>::type_t*' {aka 'short int*'}
   85 |     FORCE_INLINE static void draw(const bool sel, const uint8_t row, PGM_P const pstr, type_t * const data, ...) {
      |                                                                                        ~~~~~~~~~~~~~~~^~~~
Marlin\src\lcd\menu\menu_item.h:88:30: note: candidate: 'static void TMenuEditItem<NAME>::draw(bool, uint8_t, const char*, TMenuEditItem<NAME>::type_t (*)(), ...) [with NAME = MenuEditItemInfo_int3; uint8_t = unsigned char; TMenuEditItem<NAME>::type_t = short int]'
   88 |     FORCE_INLINE static void draw(const bool sel, const uint8_t row, PGM_P const pstr, type_t (*pget)(), ...) {
      |                              ^~~~
Marlin\src\lcd\menu\menu_item.h:88:97: note:   no known conversion for argument 4 from 'float*' to 'TMenuEditItem<MenuEditItemInfo_int3>::type_t (*)()' {aka 'short int (*)()'}
   88 |     FORCE_INLINE static void draw(const bool sel, const uint8_t row, PGM_P const pstr, type_t (*pget)(), ...) {
      |                                                                                        ~~~~~~~~~^~~~~~~
[... 61 lines cut ...]
### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Marlin bugfix-2.0.x configs - 2021-03-23.zip](https://github.com/MarlinFirmware/Marlin/files/6192506/Marlin.bugfix-2.0.x.configs.-.2021-03-23.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. 
2. 
3. 

**Expected behavior:**

`bugfix-2.0.x` should compile successfully

**Actual behavior:**

`bugfix-2.0.x` fails to compile

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

yourbuddydinec, 2021-03-23:
Just tried to compile marlin using the nightly build for a new board and running into the same problems.

labotecno, 2021-03-23:
its a bad commit revert by hand ! https://github.com/MarlinFirmware/Marlin/commit/a8ea6e646301756f138373112479713f0dfbe4dc

thisiskeithb, 2021-03-23:
Yes, CI caught it as well: https://github.com/MarlinFirmware/Marlin/actions/runs/680160469

lightmaster, 2021-03-24:
I'm not sure what that bit of code that was changed in a8ea6e6 actually does. Is there any issue with using Marlin without this commit so that it actually builds?

lightmaster, 2021-03-24:
Actually, one of the commits after the commit referenced here seems to have solved this issue.

closed 2021-03-24 by thisiskeithb