MarlinFirmware/Marlin #21429: [BUG] G29 P1 fails to deploy probe at all latest Bugfix-2.0.X SKR Mini E3 V1.2 - G28 and M48 work fine.

GhostlyCrowd, 2021-03-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bltouch probe self test works, g28 using probe as z min works
UBL G29 P1 probe doesn't deploy

### Bug Timeline

Between October 2020 and now when i moved the machine forward

### Expected behavior

TO build a mesh properly

### Actual behavior

Z raises and lowers indefinitely and the probe does not deploy

### Steps to Reproduce

G28 then G29 P1

### Version of Marlin Firmware

Bugfix 2.0.x Current as of March 23 2021

### Printer model

Ender 3

### Electronics

SKR E3 Mini V1.2 Board

### Add-ons

 Bltouch 3.1 Genuine

Attached Debug probe M114 output and configs. 

To rule out a wiring issue when the printer moves to that position i moved the tool and bet to where the G29 fails (point 1 X25.0000 Y31.0000) and ran a M48 perfectly fine

[Debug probe output.txt](https://github.com/MarlinFirmware/Marlin/files/6192829/Debug.probe.output.txt)

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6192831/configs.zip)

GhostlyCrowd, 2021-03-24:
Rolled all the way back to the last known working that was on it previously. And UBL Probing works probe deploys. SO it's definitely an issue with G29/UBL and not deploying the probe.  On current bugfix.

Wish I kept this more up to date to narrow the window more than 1000ish commits. But it's a friends printer, and he was  shy about updating as it was working. 
`C:\Users\Jeff\2.0.x-Bugfix_OCT5\Marlin>git rev-parse HEAD
343441d7465fb637f0c2627060b7003a0d0fc25c

C:\Users\Jeff\2.0.x-Bugfix_OCT5\Marlin>git rev-parse --short HEAD
343441d746

C:\Users\Jeff\2.0.x-Bugfix_OCT5\Marlin>git status
On branch bugfix-2.0.x
Your branch is behind 'origin/bugfix-2.0.x' by 1015 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   Marlin/Configuration.h
        modified:   Marlin/Configuration_adv.h
        modified:   platformio.ini

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        buildroot/share/PlatformIO/scripts/nanolib.py

no changes added to commit (use "git add" and/or "git commit -a")`

GhostlyCrowd, 2021-03-24:
Release 2.0.7.2 is working as well. Trying to step it forward a bit to narrow it down

GhostlyCrowd, 2021-03-24:
Have some time and a friend willing to let me abuse his ender 3, im walking back a week at a time so far i have made it to https://github.com/MarlinFirmware/Marlin/commit/03160719eb3d9ed31115956946826cb3b08babbc and G29 P1 no probe deploy. 

Posting for record. and info.

GhostlyCrowd, 2021-03-25:
Still working my way back

Jan 21 Version bump commit - Still broken

Dec 28 Doesnt compile, a storm of broken.

Nov 28 - Compiles and WORKS!!!!

GhostlyCrowd, 2021-03-25:
Down to Dec 29 Broken

Nov 28 working

Walking @thinkyhead "distribution updates"

thisiskeithb, 2021-03-25:
Please keep mentions to a minimum unless something is on fire. 🙂

Instead of bumping this thread with comments, it'd be better to move everything to a single post for easier tracking.

GhostlyCrowd, 2021-03-25:
100% tested several times over between https://github.com/MarlinFirmware/Marlin/commit/7226f6834e9aef295ea3a7c2599e6c5a0ebb3c10 < working G29 P1 deploys probe to https://github.com/MarlinFirmware/Marlin/commit/2dbd2063f87e8e91d5d1ebd42c027975498b29c9 <  broken Wish i could be of better help. a 4 day window is the best I can do.

GhostlyCrowd, 2021-03-25:
This is the commit that breaks Probing/UBL (and probably more cannot test as it's not  my printer and the owner doesn't like randomly driving the tool into the bed because of probe wokeness) on the SKR E3 Mini series   

Reverting this commit resolves everything. https://github.com/MarlinFirmware/Marlin/pull/20325 Fixes the SKR E3 Mini. Its 

This explains the multitude of SKR E3 users on discord all complaining about probe randomness.  A couple in the past few days we couldn't get a grasp on why it was so random.

ldursw, 2021-03-27:
To narrow down the issue could you please do some more tests?

- `platformio.ini` may be set to `STM32F103RC_btt_512K`. Try with `STM32F103RC_btt`.
- Try disabling `BLTOUCH_HS_MODE`.

Also, can you provide the debug log and a video when the probe works as expected and when it doesn't?

GhostlyCrowd, 2021-03-27:
> To narrow down the issue could you please do some more tests?
> 
> * `platformio.ini` may be set to `STM32F103RC_btt_512K`. Try with `STM32F103RC_btt`.
> * Try disabling `BLTOUCH_HS_MODE`.
> 
> Also, can you provide the debug log and a video when the probe works as expected and when it doesn't?

STM32F103RC_btt_512K. or STM32F103RC_btt made no difference he usually runs STM32F103RC_btt_512K_USB and that has the same issue, I've also verified the chip is indeed 512 with a st link previously, In my quest to resolve the issue BLTOUCH_HS_MODE had been disabled to try in my testing but i can try again.  I can see if i can get logs today from his printer. Currently, he is on the latest bug fix as of a few days ago with the commit i mentioned above reverted and all is working. 

Moving the printer off of STM32F103RC_btt_512K_USB now that its reassembled and he is using it would require him to tip over and disassemble the printer every time to flash so that's not an option for the person willing to let me test my findings on his hardware.

GhostlyCrowd, 2021-03-27:
Here is a video of it failing to deploy the probe doing a ubl and just stuck in the Probing fail loop you see i the debug logs attached to this bug report https://www.youtube.com/watch?v=EAzoWJO-_cc

ldursw, 2021-03-29:
Thanks for the details. There are a few more things that I'd like you to test if you can.

The first one is to comment out `SERIAL_PORT_2`. I'm almost sure this one won't help but we can try.

The second one has a few steps. In Configuration.h enable `#define FAN_SOFT_PWM`. Then in `src/HAL/STM32F1/timers.h` change `SERVO0_TIMER_NUM` to 1 near line 86.

[image]

I've compared the firmware files and couldn't find an obvious difference between them. Please send a known good and known bad binary preferably using the exact same configuration on both. They are located at `.pio/build/STM32F103RC_btt_512K/firmware.elf`. I need the elf file, not the bin file.

GhostlyCrowd, 2021-03-29:
@ldursw 
`#define FAN_SOFT_PWM`  Is already enabled and I believe a requirement for the SKR E3 Mini boards, Attached are the working and non working ELF files, exact same Marlin bugfix from March 25th except the working has https://github.com/MarlinFirmware/Marlin/pull/20325 reverted and thus is compiled with stsm32 @ 6.1 and Maple @ 2.1 instead of the broken one with stsm32 @ 12 and Maple @ 3.1

I will get a probing log of it working shortly. 

[elf_and_config_files.zip](https://github.com/MarlinFirmware/Marlin/files/6222152/elf_and_config_files.zip)

GhostlyCrowd, 2021-03-29:
Here is the requested Debug log of G29 P1/UBL working 
[G29_P1_Working_Debug_Log.txt](https://github.com/MarlinFirmware/Marlin/files/6222220/G29_P1_Working_Debug_Log.txt)

GhostlyCrowd, 2021-03-29:
> In Configuration.h enable `#define FAN_SOFT_PWM`. Then in `src/HAL/STM32F1/timers.h` change `SERVO0_TIMER_NUM` to 1 near line 86.
> 
> [image]

This change does indeed fix the probe deployment issues with G29 P1 so were on to something.

ldursw, 2021-03-29:
We narrowed it down to a timer issue. The only commits that look suspicious are [this one](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/73e268d703e2f22bd9f915bb088dfe8d789d493d) and [this one](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/aabca7bb09eb4f2c97b10012d5d85e73716b2463).

@rhapsodyv do you know what may be causing this issue?

rhapsodyv, 2021-03-29:
> We narrowed it down to a timer issue. The only commits that look suspicious are [this one](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/73e268d703e2f22bd9f915bb088dfe8d789d493d) and [...]

DAMMIT hahahahahahah maple change their default tone timer channel!!! 

WHHYYYYYY?!?!!? hahahah

Too many things on marlin was done with their defaults in mind. I will take a closer look.

GhostlyCrowd, 2021-03-30:
The fact that I'm not crazy and you guys are seeing this too and I can stop pulling hair out is a great feeling LOL Thanks guys!

rhapsodyv, 2021-03-31:
anyone here that have a SKR MINI E3 willing to test a PR??

GhostlyCrowd, 2021-05-31:
This was fixed in https://github.com/MarlinFirmware/Marlin/pull/21488

Essentially getting off the broken STM32F1 platform and maple  and onto a supported platform resolved this an many other small fringe bugs experienced.

closed 2021-05-31 by GhostlyCrowd