MarlinFirmware/Marlin #21435: [FR] Better file list for MKS UI/TFT_LVGL_UI

mfedoseev, 2021-03-24:
### Is your feature request related to a problem? Please describe.

The MKS LVGL UI looks nice, is themable, but the file list is almost unusable. None of my filenames fit. The two vertical lines of text under each icon fit only 12-14 characters in total, not including the ".G" at the end.

### Are you looking for hardware support?

MKS Robin Nano 2.0 / TS35

### Describe the feature you want

I think it needs to change from square tiles to a vertical list, where the file names are longer, single lines of text. Just like regular Marlin UI. And the ".g" does not add any info to the user, so it needs to go.

thinkyhead, 2021-03-24:
This feature is maintained by Makerbase.

mfedoseev, 2021-03-24:
Thank you! Submitted to https://github.com/makerbase-mks

closed 2021-03-24 by mfedoseev