MarlinFirmware/Marlin #21438: mks robin e3d 1.1 serial port error

rathorp80, 2021-03-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

#error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."

### Bug Timeline

new

### Expected behavior

compile

### Actual behavior

errors out

### Version of Marlin Firmware

bugfix-2.0.x

### Electronics

mks robin e3d 1.1

[Configuration&Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6199395/Configuration.Configuration_adv.zip)

rathorp80, 2021-03-24:
the set the ports the way the mks github said to set them up

thisiskeithb, 2021-03-24:
> SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board

You’ll need to use a different serial port. This isn’t a bug.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2021-03-24 by thisiskeithb