MarlinFirmware/Marlin #21448: [BUG] (Rumba32 F446vet6 - No pwm on ext pins PD12/13/14/15  4xch of Timer4)

studiodyne, 2021-03-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Rumba32 F446vet6 - No pwm on ext pins PD12/13/14/15  4xch of Timer4

### Version of Marlin Firmware

all

### Printer model

FANCY

Rumba32 F446vet6 - No pwm on ext pins PD12/13/14/15  4xch of Timer4
I have used ' laser PWM feature ' to try PD12/13/14/15 but no pwm too
In some variants , there is a PB6/7/8/9 Cfg for timer4 
It's too much for me 

@thinkyhead  donation if you fix it ' reasonnable donation of course '

thisiskeithb, 2021-03-25:
> thinkyhead Paypal donation if you fix it ' reasonnable donation of course '

Please keep mentions to a minimum unless something is on fire.

Someone will look at this when they can.

studiodyne, 2021-03-25:
@thisiskeithb  👍

studiodyne, 2021-03-30:
@rhapsodyv 
I believed that platform io downloaded the  vet6 variant from stm32 arduino framework git, that have  the 4 channels of TIM4 on pd12/13/14/15
I have made a new variant to have it in local directory but no effects
But If the variant is local , can you tell me what file it is , and i will work on it 
But hours after hours , I'm really sure TIM4 is already in use and not available , do you have a manipulation to list the timer in use?
Thks

rhapsodyv, 2021-03-30:
This is the updated version, that have the "full" variant: https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/RUMBA32_F446VE/PeripheralPins.c#L108
It was updated 10 months ago. Jun 7, 2020. 

Platform ship with 1.9.0 release, that is from May 2020... So, PlatformIO stm32duino is outdated...

You can try just set this in the rumba env:
```
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/refs/heads/master.zip
```

It will make you to use an updated stm32duino version, that ships with the updated variant... 

But remember one thing: analog pins from original stm32duino variant will not with Marlin right now (without a fix).

studiodyne, 2021-03-30:
@rhapsodyv  , very good job , thanks a lot , I will try this tomorow , and i will report if all is ok ,  i will check all gpios and i will if something broken 

Thanks again

studiodyne, 2021-03-31:
@thisiskeithb 
You can close , I have changed the title to easily find this fix in the base

thisiskeithb, 2021-03-31:
You have the power to close your own issues as well. Just his that Close button 🙂

closed 2021-03-31 by thisiskeithb

rhapsodyv, 2021-03-31:
@studiodyne Good to know that it fixed your issue!!

But, as I said, the default variant will have some PIN issues and you can have random behaviour.

So, I sent a PR to add a fixed variant for rumba32.

Can you test it?

https://github.com/MarlinFirmware/Marlin/pull/21497

My PayPal is https://www.paypal.com/donate?business=3WLLTNTFHK99N&currency_code=USD  ❤️

studiodyne, 2021-04-01:
15$ donation made  :smile
I have made this local variant two weeks ago, but had no effects , I will try yours and verify all is fine like the imported package

For some bad behavior pins ,  I use common pins , and analog pins/pwm are just for led , so , if sometimes I have a light special behavior , it's not a problem , and it will be visible , I will report it.
I over exploit the rumba 32 ,    3x heads / 3x heater + thermistor / one servo for switching nozzle 'by myself ' / 3 runouts motion sensors/ 1 calibration pin for the swiching nozzle auto offset 'Luzzlebot like'/2 pins for external sms alarm (pause/stop) 'i use COOLANT feature M7/M8/M9 /4 pins for RGBW led because neopixel timing is dead for more than 10 led' .

studiodyne, 2021-04-01:
For information , I use all pins ,  3 hotends/thermistors/all 6 drivers / SPI/ 3 motion sensor/1 calibration pins/ bltouch + one servo for my switching nozzle/ 2 pins for coolant 'used for sms alarm ' / 4 pins PWN for rgbw led because neopixel timing is dead if more than 10 leds/ 2 serials port , one for usb , the other for mks tft35 , all works fine

You can be sure if there is a issue in this variant , it will be visible and i will report it

rhapsodyv, 2021-04-01:
Thanks!!

This change is needed for marlin work properly: https://github.com/MarlinFirmware/Marlin/blob/ef25c7b6a07d0b45990ba51320f7023b73bae1de/buildroot/share/PlatformIO/variants/MARLIN_F446VE/variant.h#L30. `A0` to `PIN_A0`

studiodyne, 2021-04-01:
It's ok , i will update with your own variant , and forget the stm package 
I will try
Today , i have discover new tweaks and few bugs again
RGBW led , must have a 100hz pwm , if not , our eyes doesn't see the brightness change , or only if is very low
Then , RGBW led must be rewritten , to add brightness menu + Pwm frequency defined in the feature options, because ' default =20khz'
May be i'm the only one on earth to use this feature made the last century 😄 

If you have time and nothing else to do ..... 
Same donation if you write a ' set_pwm_frequency ' for RGBW led ,  with a constant defined in the feature (100 hz default) , and a brightness menu line. (Take care , the brightness must be applied on all led proportionnaly , without modifying the gradient value of each ' , like a factor that divide or multiply

If you have time and nothing else to do
Same donation again if you create a ' STARTUP ANIMATION' , with and without pwm 
With PWM '  you fade between colors set in the user presets '(option of speed in the feature) for better effects will need ' pwm min and pwm max for the animation )
No pwm , you make some flickering animation (speed option)

If not time or not interested i will make it by myself
THANKSSSSSSSSSSSSS FOR ALLLL