MarlinFirmware/Marlin #21449: [BUG] AUTO_POWER_CONTROL    does not respect    PSU_POWERUP_DELAY 500

Flutschi84, 2021-03-25:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When i turn on the Heatbed via my  Reprap Full Graphics Display, the onboard Heatbed-Mosfet instantly turns on. 
But the PSU is powered on delayed by ~1s. And i'm talking about the logical Pin that PSU is connectet to. NOT the time that the 12V arrives at my 12V Input.

And now my Problem with this behaviour:
Because of TMC2130 and TMC2209 internal design, i have to put a Shotky Diod from Logical TMC Power to Motor Power ( e.g. to my 12V rail).
When My Printer is idle, it is powered via external Meanwell 5V Supply (not via USB). and die 5V goes throuh the Protection Diod at TMC's into my "not powered" 12V Rail. Because of ForwardVoltage of that Diod there are now about 4,6V on my 12V input Rail. 

But when now the Heatbed Mosfet turns on, without the PSU is turned on, my 5V supply is massiv overloaded. Sometimes it is just enough to let the onboard 3,3V regulator to drop and the CPU is performing a reboot of the whole printer. In the other times, only the 5V is overloaded, the LCD Background is getting nearly dark, but after a time of ~1s the psu is turned on and everything is fine.

OK, someone could say i shouldnt wire things like this on my MKS SGEN_L, but the Diod at the TMC is highly recommended by Trinamic. And isn't it not an bug when the PSU is turned on AFTER the heavyest load in the whole System is turned on?

Here some relevant lines of my conf:

/**
 * Power Supply Control
 *
 * Enable and connect the power supply to the PS_ON_PIN.
 * Specify whether the power supply is active HIGH or active LOW.
 */
#define PSU_CONTROL
#define PSU_NAME "Server-PSU"
#define PS_ON_PIN E1_DIAG_PIN         // Define the Pin that enables the PSU
#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH       // Set 'LOW' for ATX, 'HIGH' for X-Box
  
  #define PSU_DEFAULT_OFF             // Keep power off until enabled directly with M80
  #define PSU_POWERUP_DELAY 500       // (ms) Delay for the PSU to warm up to full power
  
  #define AUTO_POWER_CONTROL          // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 1
  #endif
#endif

### Bug Timeline

I dont know

### Expected behavior

PSU is turned on 500ms before Heatbed turns on.

### Actual behavior

Mosfet for Bed Heater Turns on about 1s before PSU is started. (I mean logicly turned on at my PSU pin. I know that my Server PSU need som ms to deliver Power. And therefore i have configured 500ms delay.)
By the Way: when i home axis on the idle Printer, the delay works correct. i see instantly 3.3 V @ PSU pin an AFTER the configured 500ms delay, the motors engage.

### Version of Marlin Firmware

2.0.7.2

### Printer model

Heavy modif Geeetech i3 ProB

### Electronics

MKS SGEN_L 32bit , 3x TMC2130 2x TMC2209

### Add-ons

Full Graphic RepRap Dicscount Display

### Your Slicer

Slic3r

### Host Software

Repetier Host

thisiskeithb, 2021-03-25:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Flutschi84, 2021-03-26:
Good Morning,

i have tested with latest Bugfix Branch, and it is still the same behaviour.

I've attached my configs that i converted for Bugfix Branch...

Many Thanks !!

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6210601/Configs.zip)

closed 2021-06-05 by github-actions[bot]