MarlinFirmware/Marlin #21455: [BUG] Filament Change with nozzle temperature change deadlocks.

BigJBehr, 2021-03-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.7.2, Marlin 2.0.x BugFix,  CONFIGURATION_H_VERSION 020008

Marlin 2.0 has an issue with nozzle temperature change and the M600 filament change command. If the nozzle needs to increase temperature then Marlin stalls waiting for the nozzle temperature to reach the new setpoint. However, it never does. It will maintain the previous temperature, but will never reach the new setpoint, so it never moves on. The LCD shows the current temperature and the desired temperature, but no effort is made to achieve the desired temperature.

To reproduce this problem;
Edit any gcode file. Find a layer change. Add an M104 Sxxx command to increase the nozzle temperature by 10 and add an M600 command to change the filament. The M104 command should precede the M600 command. The printer will pause for the filament change, but will never complete the operation because the new nozzle temperature will never be reached. Nozzle heating continues with the old nozzle setpoint even though the LCD displays the new nozzle setpoint. The user cannot exit the filament change operation (the button & rotary encoder do not function). Your only choices are to reset or power off.

If an M109 (wait for temperature change) is used then the print head will stop and hover at the last layer height while the nozzle heats up to the new temperature. The new nozzle setpoint is displayed on the LCD. The M600 command is never executed because the nozzle temperature is held at the previous temperature and no attempt is made to reach the new setpoint.

### Bug Timeline

First noticed 3-2021

### Expected behavior

I expected the nozzle temperature change to occur either prior to (M109) or during (M104) the M600 filament change operation.

### Actual behavior

System stalls waiting for nozzle temperature change when changing filament. Filament change never finished due to nozzle never changing temperature.

### Steps to Reproduce

To reproduce this problem;
Edit any gcode file. Find a layer change. Add an M104 Sxxx command to increase the nozzle temperature by 10 and add an M600 command to change the filament. The M104 command should precede the M600 command. The printer will pause for the filament change, but will never complete the operation because the new nozzle temperature will never be reached. Nozzle heating continues with the old nozzle setpoint even though the LCD displays the new nozzle setpoint. The user cannot exit the filament change operation (the button & rotary encoder do not function). Your only choices are to reset or power off.

### Version of Marlin Firmware

Marlin 2.0.7.2, Marlin 2.0.x BugFix,  CONFIGURATION_H_VERSION 020008

### Printer model

Prusa MK2.5S with Btt SKR 1.3 motherboard with Marlin 2.0

### Electronics

Btt SKR 1.3 motherboard with Marlin 2.0

### Your Slicer

Cura

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6213897/Configs.zip)

Marlin 2.0.7.2, Marlin 2.0.x BugFix,  CONFIGURATION_H_VERSION 020008

Marlin 2.0 has an issue with nozzle temperature change and the M600 filament change command. If the nozzle needs to increase temperature then Marlin stalls waiting for the nozzle temperature to reach the new setpoint. However, it never does. It will maintain the previous temperature, but will never reach the new setpoint, so it never moves on. The LCD shows the current temperature and the desired temperature, but no effort is made to achieve the desired temperature.

To reproduce this problem;
Edit any gcode file. Find a layer change. Add an M104 Sxxx command to increase the nozzle temperature by 10 and add an M600 command to change the filament. The M104 command should precede the M600 command. The printer will pause for the filament change, but will never complete the operation because the new nozzle temperature will never be reached. Nozzle heating continues with the old nozzle setpoint even though the LCD displays the new nozzle setpoint. The user cannot exit the filament change operation (the button & rotary encoder do not function). Your only choices are to reset or power off.

If an M109 (wait for temperature change) is used then the print head will stop and hover at the last layer height while the nozzle heats up to the new temperature. The new nozzle setpoint is displayed on the LCD. The M600 command is never executed because the nozzle temperature is held at the previous temperature and no attempt is made to reach the new setpoint.

thisiskeithb, 2021-03-26:
If you add an [`M400` (Finish Moves)](https://marlinfw.org/docs/gcode/M400.html) after `M104 S***`, but before `M600`, does it work correctly?

BigJBehr, 2021-03-27:
Adding an M400 command after the M104 Sxxx or M109 Sxxx command and before the M600 command does not solve the issue. However, using the M109, M400 sequence allows the user to access the menu system. Going to the “Tune, Nozzle” menu allows you to enter a new temperature setpoint. However, no attempt is made to actually change the nozzle temperature. “Stop Print” will abort the print in progress.

The M104, M400 sequence does not allow menu system access.

Sebazzz, 2021-03-31:
It is something in the DOGM LCD code. On an ExtUI implementation of M600 filament change, this issue doesn't occur and I was able to resume printing just fine.

BigJBehr, 2021-05-31:
So am I the only person printing in two colors on a single extruder system ? This works fine on my Prusa, but not on my printer running Marlin 2.0.

samwiseg0, 2021-08-01:
Bump to prevent closing.

ellensp, 2021-08-01:
Has anyone tried this on current bugfix? Its been a while...

thisiskeithb, 2021-09-24:
This bug may still exist, but note that the `M600` command now supports the `R` parameter to set the resume temperature: https://github.com/MarlinFirmware/Marlin/pull/17919

closed 2021-12-12 by github-actions[bot]