MarlinFirmware/Marlin #21456: [BUG] G26 does not respect MANUAL_Y_HOME_POS

strayr, 2021-03-27:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I've just fitted a gantry adapter that moves my hotend and probe - 7mm in Y. I could M206 but i set it in firmware with #define MANUAL_Y_HOME_POS = -7 to make sure i kept track of where (maybe) corner screws and suchlike were. It appears to be measuring and editing the UBL mesh in the right place, 0,0 is on the print bed, but the Y=0 points fall off the end of the print bed when printing a G26. This could be me doing a stupid but everything else seems to work. 

### Expected behavior

G26 fits on the bed and aligns with the mesh points when cycling through mesh edit on the LCD

### Actual behavior

G26 pattern starts from the endstop position 7mm away and the y=0 row falls off the ned of the bed, points don't match the mesh edit lcd function.

### Steps to Reproduce

#define MANUAL_Y_HOME_POS = -7
#define Y_MIN_POS -7 // set software endstop to match the physical endstop

make, install firmware etc
G29 P1
G29 P3
G29 P5C
G29 A
G29 S0

G26

### Version of Marlin Firmware

2.0.6.1 bugfix

### Printer model

ender3 ~v2 

### Electronics

4.2.2 mainboard

### Add-ons

bltouch, offset hotend gantry

### Host Software

OctoPrint

thisiskeithb, 2021-03-27:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, please attach your configs in a single `.zip`.

strayr, 2021-03-27:
It's fixed in bugfix-2.0.x
as is the issue that had me using an odd version that kinda worked, so time to mark some more bugs fixed
thanks everyone for making marlin so awesome

closed 2021-03-27 by strayr