MarlinFirmware/Marlin #21459: Screen garbled for Tevo Tornado using latest Firmware

rajinder-yadav, 2021-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After update my Tevo Tornado with latest firmware, the screen is unreadable and messy.

My configuration file is below:

https://github.com/rajinder-yadav/Tevo-Tornado-Configuration

### Bug Timeline

Using latest firmware

### Expected behavior

readable screen

### Actual behavior

screen is unreadable

### Steps to Reproduce

build 
update firmware

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.0.7.2 

### Printer model

Tevo Tornado V2

### Electronics

standard motherboard

### Add-ons

TMC2209

ellensp, 2021-03-27:
its very unusual for a mega2560 based board to need this... must be a really slow lcd.

add these to your configuration.h
```
  #define BOARD_ST7920_DELAY_1 DELAY_NS(250)
  #define BOARD_ST7920_DELAY_2 DELAY_NS(250)
  #define BOARD_ST7920_DELAY_3 DELAY_NS(250)
```

closed 2021-03-27 by ellensp

rajinder-yadav, 2021-03-28:
Thanks this worked.