MarlinFirmware/Marlin #21460: [FR] (Turn fan on while heater on)

wdq4587, 2021-03-27:
### Describe the feature you want

My 3D printer is a simple 3D printer only has one fan using +12V power supply. I did not like it's noise so I change it to RAMPS 1.4 board D8 pin and now fan run only after temperature > 50.  But I heard it's not safe if sensor failed. So my suggestion is change the logic to if heater on or temperature > 50, then fan  run. I suppose that will reduce the dangerous getting  fire. I am using Marlin 1.1.9.1

ellensp, 2021-03-27:
Marlin 1 is no longer being worked on.
Marlin 2 has thermal runaway protection, if hotend is on and temperature is not rising, the printer is halted

You should upgrade to marlin 2, yes it is for ramps boards also.

closed 2021-03-27 by ellensp

wdq4587, 2021-03-30:
I have try Marlin 2.0.7.2，but I get link error "CreateProcess: No such file or directory"
But after what you saying, I found this: 
https://forum.arduino.cc/?topic=721878#msg4936118
And today finally upgrade to 2.0.7.2