MarlinFirmware/Marlin #21461: [BUG] MKS robin nano v.2 - the temperature on the main screen is not updating

Malderin, 2021-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating https://github.com/MarlinFirmware/Marlin/commit/532df198da6a9cadb441a1993b4b482e69e317ce, bed temperature and hotend temperature(+% fan)  have been added to the start screen.
When heated, the bed and hotend temperatures are not updated. Temperatures frozen (unchanged).

### Version of Marlin Firmware

2.0.x bugfix

thinkyhead, 2021-03-29:
A patch has been merged. If it solves the issue then this can be closed!

Malderin, 2021-03-29:
@rhapsodyv ,  
Something is wrong after the update
[image]

DmitryAranax, 2021-03-29:
To update a bed temperature, hotend temperature and a fan percent, you need a fix code at draw_ready.cpp, draw_ready.h and draw_ui.cpp (GUI_RefreshPage() -> case PRINT_READY_UI). Original commit can only show parameters once not update automatically. To update them you need an additional routine like others update routines in MKS UI. Check files please.
[draw_ready.zip](https://github.com/MarlinFirmware/Marlin/files/6221649/draw_ready.zip)

Malderin, 2021-03-29:
> To update a bed temperature, hotend temperature and a fan percent, you need a fix code at draw_ready.cpp, draw_ready.h and draw_ui.cpp (GUI_RefreshPage() -> case PRINT_READY_UI). Original commit can [...]

look at this https://github.com/MarlinFirmware/Marlin/pull/21468

DmitryAranax, 2021-03-29:
Indeed. Great work!

closed 2021-04-06 by thinkyhead