MarlinFirmware/Marlin #21462: [BUG] TMC2209 UART current not working on MKS Robin Nano V1.2

Adi146, 2021-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I´m using FYSETC TMC2209 V3.1 Drivers on a MKS Robin Nano V1.2 with UAR. I can set the driver current but it seems to do nothing.

### Expected behavior

Motor salling when using low current.

### Actual behavior

I can set the current to 100 mA but the motor is still moving normally. Everything else like Stallguard and the switch to Spreadcycle works as expected. I tested several pins for UART communication

### Steps to Reproduce

1. Set the current to very low value
2. Try to move the printhead

### Version of Marlin Firmware

2.0.x bugfix

### Printer model

FLSUN Q5, Two Trees Sapphire Plus

### Electronics

MKS Robin Nano V1.2, FYSETC TMC2209 V3.1

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6216061/Configuration.zip)

ellensp, 2021-03-28:
MKS Robin Nano V1.2  does not have UART capability.
There are no uart tracks on the PCB

ellensp, 2021-03-28:
you need to modify the controller and the pins files.

Adi146, 2021-03-28:
Yes I tried the wifi pins and the pins of a spare stepper slot with same results. What I forgot to mention is that M122 reports the correct settings.

Adi146, 2021-03-30:
I just have boards with the VET6 processor so I always used the mks_robin_nano35 env. But I just flashed a mks_robin_nano35_stm32 marlin which somehow also works and the stepper skipped as it should. Maybe it was my fault after all that I used the wrong env or do I miss something?

rhapsodyv, 2021-03-30:
It’s not your fault.

There’re two envs for nano v1 and v2 board. One uses the old framework maple. And the new env, uses a new framework stm32duino.

maple (mks_robin_nano35) will be discontinued soon. The new one (mks_robin_nano35_stm32) will be the default.

If your issue is solved when using the new env, that great and you keep using it, and you can close this.

Our objective is avoid fixing issues on old env, because it will be discontinued.

closed 2021-03-30 by Adi146