MarlinFirmware/Marlin #21464: TMC2209 V2 reporting error after update on Tevo Tornado

rajinder-yadav, 2021-03-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After upload latest firmware build I am getting error report got my TMC2209 v2 driver

### Bug Timeline

New bug, Using latest firmware

### Expected behavior

TMC2209 driver should report no error

### Actual behavior

Send: M122
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW

### Steps to Reproduce

Use gcode M122

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.0.7.2

### Printer model

Tevo Tornado V2

### Electronics

standard motherboard

### Add-ons

TMC2209

### Your Slicer

Cura

### Host Software

OctoPrint

Here is the following configuration: https://github.com/rajinder-yadav/Tevo-Tornado-Configuration

closed 2021-03-28 by ellensp

ellensp, 2021-03-28:
This is always a wiring or configuration issue. not a marlin bug.

rajinder-yadav, 2021-03-28:
Well this was not an problem with the previous Marlin firmware I had built and installed several weeks back. Using the same configurations settings and diffing files I've verified the configuration files changes and don't see anything off.

This showed up after using the latest firmware, so I see this as a legit issue.

rhapsodyv, 2021-03-28:
If you switch now, for that old marlin version you said, everything works?

rajinder-yadav, 2021-03-28:
@rhapsodyv so I think I know what I did, my board MKS Gen L V1.0 doesn't support UART mode, let me know if my fix make sense. What I ended up doing was comment out the drivers in Configuration.h

```cpp
// #define X_DRIVER_TYPE TMC2209
// #define Y_DRIVER_TYPE TMC2209
// #define Z_DRIVER_TYPE TMC2209
// #define E0_DRIVER_TYPE TMC2209
```

Am I correct to assume, the above line control UART mode?