MarlinFirmware/Marlin #21465: [BUG] FILAMENT_RUNOUT_SENSOR_DEBUG causes marlin to hang

samwiseg0, 2021-03-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In trying to debug my filament sensor being tripped randomly I tried to enable `FILAMENT_RUNOUT_SENSOR_DEBUG` in `runout.h` and it causes marlin to hang. I do have `WATCHDOG_DURATION_8S` enabled. I am not sure if this is related but also trying to perform a change filament from the LCD causes the printer to hang as well. Lastly, Doing a filament unload or load does show the appropriate debug output in the terminal. 

Shorty after a print job is sent to print the printer hangs after the gcode below. The printer never makes it to the intro line. 
```gcode
M104 S215 ; set extruder temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
```

Video of the hang can be viewed here. 
https://www.youtube.com/watch?v=Lz7LM_cap2A

I have also tried printing from the SD card and it does start the print but I am unable to maintain a serial connection to the printer which defeats the purpose of the debug I am trying to perform. 

### Bug Timeline

Unknown. I just enabled this for the first time.

### Expected behavior

Marlin not to hang and print the runout sensor debug to the terminal.

### Actual behavior

The printer hangs and restarts. 

### Steps to Reproduce

1. Compile at commit https://github.com/MarlinFirmware/Marlin/commit/5991836e1fdd73f7a668cfc4221bfe3017a2580d
2. Start a print from a serial connection
3. Observe the printer hangs when heating the hotend and bed. 

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 27 2021 23:18:40) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-5 Pro 4.2.2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff Cap:SERIAL_XON_XOFF:0 Cap:BINARY_FILE_TRANSFER:0 Cap:EEPROM:1 Cap:VOLUMETRIC:1 Cap:AUTOREPORT_TEMP:1 Cap:PROGRESS:0 Cap:PRINT_JOB:1 Cap:AUTOLEVEL:1 Cap:RUNOUT:1 Cap:Z_PROBE:1 Cap:LEVELING_DATA:1 Cap:BUILD_PERCENT:1 Cap:SOFTWARE_POWER:0 Cap:TOGGLE_LIGHTS:0 Cap:CASE_LIGHT_BRIGHTNESS:0 Cap:EMERGENCY_PARSER:1 Cap:PROMPT_SUPPORT:1 Cap:SDCARD:1 Cap:REPEAT:0 Cap:SD_WRITE:1 Cap:AUTOREPORT_SD_STATUS:0 Cap:LONG_FILENAME:0 Cap:THERMAL_PROTECTION:1 Cap:MOTION_MODES:0 Cap:ARCS:1 Cap:BABYSTEPPING:1 Cap:CHAMBER_TEMPERATURE:0 Cap:COOLER_TEMPERATURE:0 Cap:MEATPACK:0

### Printer model

Ender 5 Pro v4.2.2

### Electronics

Stock

### Add-ons

None

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

`Configuration.h` and `Configuration_adv.h`
https://gist.github.com/samwiseg0/2cd33e76c2709b24e38053c9f2fd2729

Sample gcode file that will crash (I have tried several files and they all exhibit the same behavior. It does not seem to matter what gcode file you try and print):
[self_watering_0.2mm_PLA_T210_B60_ENDER5_23h31m.arcwelded.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/6216852/self_watering_0.2mm_PLA_T210_B60_ENDER5_23h31m.arcwelded.gcode.zip)

gx1400, 2021-04-06:
Same issue occurred on my Creality CR-6 SE

Configuration.h: https://pastebin.com/8Ax93vpR
Configuration_adv.h: https://pastebin.com/54qhVvrq

I have a Creality CR-6 SE with a BTT CR6 SKR v1.0 motherboard using stock Creality TFT.  Using the Marlin fork from https://github.com/CR6Community/Marlin (built on commit dd0a515096 (HEAD -> extui))

I've been trying to connect up the BTT Smart Filament Sensor but Marlin throws a runout alarm every time the encoder triggers a runout condition, so I was attempting to enable debug in runout.h so I could see what is happening on the serial interface.

I uncommented "//#define FILAMENT_RUNOUT_SENSOR_DEBUG" on Line 41 of runout.h.  

I'm print using Octoprint on a Raspberry Pi 4 4GB.  When I start a job, the serial interface connection in Octoprint is immediately disconnected.

samwiseg0, 2021-04-06:
> Same issue occurred on my Creality CR-6 SE
> 
> Configuration.h: https://pastebin.com/8Ax93vpR
> Configuration_adv.h: https://pastebin.com/54qhVvrq
> 
> I have a Creality CR-6 SE with a BTT CR6 SKR v1.0 motherboard using stock Creality TFT. Using the Marlin fork from https://github.com/CR6Community/Marlin (built on commit [dd0a515](https://github.com/MarlinFirmware/Marlin/commit/dd0a515096083fc5eb2a1e649d22b47ff43653e7) (HEAD -> extui))
> 
> I've been trying to connect up the BTT Smart Filament Sensor but Marlin throws a runout alarm every time the encoder triggers a runout condition, so I was attempting to enable debug in runout.h so I could see what is happening on the serial interface.
> 
> I uncommented "//#define FILAMENT_RUNOUT_SENSOR_DEBUG" on Line 41 of runout.h.
> 
> I'm print using Octoprint on a Raspberry Pi 4 4GB. When I start a job, the serial interface connection in Octoprint is immediately disconnected.

Yep it is still a problem. I have periodically recompiled in hopes there is a fix but it is still broken. I hope someone can look at it soon as it would be nice to rule out my sensor or marlin causing a random runout condition.

samwiseg0, 2021-06-01:
Bump to prevent stalebot actions.

Anathema-Device, 2021-06-22:
I have been trying to get this to work as well on a mega2560 board. 2.0.8 sorta works if I set the runout length to be 10mm rather than the recommended 7mm but it fails right off with 2.0.9.  The current (post 2.0.9) bugfix seems to be back with the 2.0.8 behaviour.  The problem is worse with the runout debug turned on so I think that it may be related to performance?  That is, a change may be reported by the sensor but the runout code does not look before triggering the runout.  I put lights on the runout sensors and I see it changing state just fine and it appears that when a false runout is triggered it is shortly after a state change on the sensor which would appear to indicate to me that the board is not picking up the change and assuming the runout state.  If there was a check for a changed state prior to triggering the M600 action this may be avoided?  How can I debug this and is there any docs about how the code should run that I can browse?  Thanks.

samwiseg0, 2021-06-22:
> if I set the runout length to be 10mm rather than the recommended 7mm but it fails right off with 2.0.9. The current (post 2.0.9) bugfix seems to be back with the 2.0.8 behaviour. 

I have found that this is inconsistent at best. I was able to run for a time with 7 then 10 and at this point I am up to 20 and that is the only value I have been able to run for months without issue. 

> I put lights on the runout sensors and I see it changing state just fine and it appears that when a false runout is triggered it is shortly after a state change on the sensor which would appear to indicate to me that the board is not picking up the change and assuming the runout state.

What I have done is in my end gcode disable the runout sensor and then in my start gcode, after my intro line, enable the sensor once more. This coupled with 20mm runout distance has worked around the issue so far.

I am fairly confident that there is a bug in marlin with "smart sensors" but I would like to enable the debug in marlin to see what it thinks. Hopefully once this bug is fixed I can gather that data point.

> If there was a check for a changed state prior to triggering the M600 action this may be avoided?

Optimally the false triggering should be fixed vs some other workaround such as this. 

> How can I debug this and is there any docs about how the code should run that I can browse?

As far as I know. There are no docs for this. At least I have not been able to find any. I think the best we can do is get the debug from marlin but that is broken at the moment.

Anathema-Device, 2021-06-23:
Thanks @samwiseg0 those are good suggestions, I will give those changes a try - easy to do and it looks like a good chance for joy ;)

Just to add a bit of information to this from more recent observations that I have made and maybe let someone more knowing poke in the right places. Originally I was running with Meatpack on, runout debug messages on, and the gcode run through Arc Welder.  With that I was getting communications state errors so I turned off Meatpack and the failure was only on displaying some odd huge value  so I tried a couple of versions of Marlin and found that 2.0.9 was the worst. 
Now running post-2.0.9 2.0.bugfix.  Turning off Arc Welder now seems to have made a huge difference - I can get almost through a layer without the runout sensor being detected as bad so I now guess that it has to do with some bad 
calculation but I'm still not seeing where it might be in the code - mostly because I don't understand its operation well enough.

Running the printer with Octoprint driving Marlin 2.0.8 with Meatpack off this is what the typical failure looks like.  Notice the bad value that causes the run-out. 
```
Recv: Motion detected: 0
Recv: Remaining mm: 7.00, 10.00
Recv: T:231.00 /230.00 B:74.96 /75.00 @:32 B@:47
Recv: Motion detected: 0
Recv: Remaining mm: 5.26, 10.00
Recv: Runout Sensors: 4948484848484848 -> 0 RUN OUT
Recv: Runout Command: M600
Recv: T:230.06 /230.00 B:74.95 /75.00 @:43 B@:50
Recv: echo:busy: processing
```
So, I think it is likely just some simple calculation error.

samwiseg0, 2021-08-22:
Bump for stale bot

samwiseg0, 2022-02-22:
This will be my last comment on this issue. If someone else would like to keep this one alive please comment the next time stale bot comes around.

closed 2022-05-06 by github-actions[bot]