MarlinFirmware/Marlin #21466: Robin Nano 1.2 with UART and Neopixel not working

galluccib, 2021-03-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Have MKS Robin Nano 1.2 with MKS 2209 V2 running Marlin current bugfix and when in standalone NeoPixels are working fine on pin PB2

### Bug Timeline

new

### Expected behavior

For the NeoPixel to work.

### Actual behavior

After going from standalone to UART it stopped working.

### Steps to Reproduce

Go from standalone to uart.

### Version of Marlin Firmware

Current Bugfix as of 3/27

### Printer model

MKS Robin Nano 1.2

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

Have MKS Robin Nano 1.2 with MKS 2209 V2 running Marlin current bugfix and when in standalone NeoPixels are working fine on pin PB2. After switching to UART they no longer work. If I go back to standalone they started working. Also tried switching to pin PC4 but that didn't work either. Only thing added to the pins file is below and configs are attached. Using five steppers so using PA5 for the fifth one.

#if HAS_TMC_UART

#define X_SERIAL_TX_PIN PA10
#define X_SERIAL_RX_PIN PA10

#define Y_SERIAL_TX_PIN PA9
#define Y_SERIAL_RX_PIN PA9

#define Z_SERIAL_TX_PIN PC7
#define Z_SERIAL_RX_PIN PC7

#define E0_SERIAL_TX_PIN PC13
#define E0_SERIAL_RX_PIN PC13

#define E1_SERIAL_TX_PIN PA5
#define E1_SERIAL_RX_PIN PA5

// Reduce baud rate to improve software serial reliability
#define TMC_BAUD_RATE 19200
#endif // HAS_TMC_UART

galluccib, 2021-03-28:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6216872/Configuration.zip)

rhapsodyv, 2021-03-28:
Nano V1.2 doesn't support TMC on UART mode: https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X 

```
————————————————————————————————————————————————————————————————————————————————————————————————————————
                        MKS Robin Nano V1.x                              MKS Robin Nano V2.0            
————————————————————————————————————————————————————————————————————————————————————————————————————————
Firmware support        MKS Robin Nano Firmware/Above Marlin2.0.6        Above Marlin2.0.6   
————————————————————————————————————————————————————————————————————————————————————————————————————————
LCD Type and Interface  MKS Robin TFT35(FSMC)                            MKS TS35(SPI),LCD12864,LCD2004    
————————————————————————————————————————————————————————————————————————————————————————————————————————
Rotary Encoder          Not support                                      Support  
————————————————————————————————————————————————————————————————————————————————————————————————————————
Motor Driver mode       STEP/DIR                                         STEP/DIR,TMC-UART,TMC-SPI    
————————————————————————————————————————————————————————————————————————————————————————————————————————
On board EEPROM         2KB                                               4KB                          
————————————————————————————————————————————————————————————————————————————————————————————————————————
```

galluccib, 2021-03-28:
https://youtu.be/7ShcFKXrVHo

ellensp, 2021-03-28:
reopened...   is a "hack" to get uart working.

willngton, 2021-03-28:
I have tested with a build from toda'y bug-fix and could not simulated the issue.

Lower X current to 100 and was able to finger hold it while trying to move left or right.

Configuration is the same as posted above, however I have only E0

rhapsodyv, 2021-03-28:
I can't even build your configs.

First I get `NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!`. Then I commented.

So I get: `Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:2255:6: error: #error "NEOPIXEL_LED requires NEOPIXEL_PIN and NEOPIXEL_PIXELS."`. Then I changed it to `PB2` instead of LED_PIN.... See `LED_PIN` IS NOT NEOPIXEL_PIN... It will be used in the code to turn on and of the LED!! It should not be used as a config for NeoPixel led.

Then I get this: 
```
*** [.pio/build/mks_robin_nano35/src/src/MarlinCore.cpp.o] Error 1
In file included from Marlin/src/feature/leds/neopixel.h:34:0,
                 from Marlin/src/feature/leds/leds.h:33,
                 from Marlin/src/feature/leds/printer_event_leds.h:28,
                 from Marlin/src/feature/leds/printer_event_leds.cpp:31:
.pio/libdeps/mks_robin_nano35/Adafruit NeoPixel/Adafruit_NeoPixel.h:361:3: error: 'GPIO_TypeDef' does not name a type
   GPIO_TypeDef *gpioPort;       ///< Output GPIO PORT
   ^~~~~~~~~~~~
*** [.pio/build/mks_robin_nano35/src/src/feature/leds/printer_event_leds.cpp.o] Error 1
```

In resume:

1) You should not use `LED_PIN` with neopixel. You should use direct the PIN you want (like PB2).

2) You need provide configurations that generate a successful build. If you changed `platformio.ini`, you should provide it too.

willngton, 2021-03-28:
Comparing your config with mine, I found out that is missing on yours adv.h SLAVE_ADDRESS  for used UART pins,

Could you test with it enable ?

 *
   * Set *_SERIAL_TX_PIN and *_SERIAL_RX_PIN to match for all drivers
   * on the same serial port, either here or in your board's pins file.
   */
  //#define  X_SLAVE_ADDRESS 0
  //#define  Y_SLAVE_ADDRESS 0
  //#define  Z_SLAVE_ADDRESS 0
  //#define X2_SLAVE_ADDRESS 0
  //#define Y2_SLAVE_ADDRESS 0
  //#define Z2_SLAVE_ADDRESS 0
  //#define Z3_SLAVE_ADDRESS 0
  //#define Z4_SLAVE_ADDRESS 0
  //#define E0_SLAVE_ADDRESS 0
  //#define E1_SLAVE_ADDRESS 0
  //#define E2_SLAVE_ADDRESS 0
  //#define E3_SLAVE_ADDRESS 0
  //#define E4_SLAVE_ADDRESS 0
  //#define E5_SLAVE_ADDRESS 0
  //#define E6_SLAVE_ADDRESS 0
  //#define E7_SLAVE_ADDRESS 0

galluccib, 2021-03-28:
Are you using the old way of doing UART prior to MKS releasing the video showing the pins to use? From what I understood was the - #define X_SLAVE_ADDRESS - was used when mapping all five steppers to two pins.

galluccib, 2021-03-28:
> I can't even build your configs.
>
> First I get `NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!`. Then I commented.
>
> So I get: [...]

Had zero issues using LED_PIN as the pins file #define LED_PIN - > PB2.. Again if I go back to standalone from UART it works fine.

> I can't even build your configs.
>
> First I get `NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!`. Then I commented.
>
> So I get: [...]

galluccib, 2021-03-28:
> I can't even build your configs.
>
> First I get `NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!`. Then I commented.
>
> So I get: [...]

I have zero build issues.. I have been using the - LED_PIN for months now with zero issues. Pins file "#define LED_PIN PB2". Again with it in standalone mode works fine its just when switching to UART it stops sending data for the pixel.

galluccib, 2021-03-28:
> Comparing your config with mine, I found out that is missing on yours adv.h SLAVE_ADDRESS for used UART pins,
>
> Could you test with it enable ?
>
>     * * Set *_SERIAL_TX_PIN and *_SERIAL_RX_PIN to [...]

Are you using the old way of doing UART prior to MKS releasing the video showing the pins to use? From what I understood  the - #define X_SLAVE_ADDRESS - was used when mapping all five steppers to two pins.

galluccib, 2021-03-28:
pins file - When I asked the MKS folks they said to use PA5 for the fifth *(Z Stepper)

#if HAS_TMC_UART

  #define X_SERIAL_TX_PIN                   PA10
  #define X_SERIAL_RX_PIN                   PA10

  #define Y_SERIAL_TX_PIN                   PA9
  #define Y_SERIAL_RX_PIN                   PA9

  #define Z_SERIAL_TX_PIN                   PC7
  #define Z_SERIAL_RX_PIN                   PC7

  #define E0_SERIAL_TX_PIN                  PC13
  #define E0_SERIAL_RX_PIN                  PC13
  
  #define E1_SERIAL_TX_PIN                  PA5
  #define E1_SERIAL_RX_PIN                  PA5

  // Reduce baud rate to improve software serial reliability
  #define TMC_BAUD_RATE                    19200
#endif // HAS_TMC_UART

galluccib, 2021-04-02:
I dumped the .pio directory and pulled a current branch and seem to be working using mks_robin_nano35 (Which is what I've always used). Odd because it doesn't looking like there was any updates or changes for the Robin board and when I was building before never saw any errors... Very odd....

But yes it's working now..

rhapsodyv, 2021-04-02:
So this can be closed?

galluccib, 2021-04-02:
> 
> 
> So this can be closed?

Yes...

closed 2021-04-02 by galluccib