MarlinFirmware/Marlin #21471: [BUG] Random/unintended tool change during layer print and in idle! (was: #21022)

dskw, 2021-03-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The printer has two fixed hotends with 16.5 mm X-axis offset. They use mechanically identical extruders and no switching via servo or whatsoever.
As described in the old issue #21022 the printer suddenly switches tool heads during a print.
With the latest bugfix-2.0.x branch this now also happens during idle!

When printing a model using a single toolhead I noticed that the firmware even correctly reports "cold extrusion prevented" after one of these unintended tool head changes happens mid print.

Please advice if I did something wrong on the configuration as I don't see where this behavior might be coming from.

### Bug Timeline

#21022

### Expected behavior

No toolhead change unless a T<#> gcode has been issued.

### Actual behavior

The printer executes a tool head change during print and in idle, even though only a single T0 is present at the start of the gcode file.

### Steps to Reproduce

slice the attached flat_test.stl (gcode from Simplify3D included)
start the print
wait until a random tool change happens

### Version of Marlin Firmware

bugfix-2.0.x (c45b91aa94c7008e3fd8ea297df57948af9158a3)

### Printer model

TwoTrees Sapphire Pro

### Electronics

MKS Robin Nano 2.0

### Add-ons

DForce Spider Hotend (twice)

### Your Slicer

Simplify3D

### Host Software

OctoPrint

[Marlin-bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/6217638/Marlin-bugfix.zip)

At first I thought it could be a hardware error, but the "cold extrusion prevented" messages and the exact movement matching a regular toolhead change makes me think this is rather a software bug.

Please advice on how to debug this further.
Is there any way to modify the firmware for more debugging output during tool changes? Any way to print a simple stacktrace through the serial connection during a toolchange?

dskw, 2021-03-29:
[Marlin_bugfix_lessdiffs.zip](https://github.com/MarlinFirmware/Marlin/files/6221837/Marlin_bugfix_lessdiffs.zip)
I have updated the configuration to have less differences to the submitted default config.
Still have the issues though.

I'm starting to dig into how to use the debug output of marlin with the tool change.
Maybe I an add something to output a flag when the change was triggered by a T# command or not...

ellensp, 2021-03-29:
what build env are you using mks_robin_nano35_stm32 or mks_robin_nano35 ?

dskw, 2021-03-30:
default_envs = mks_robin_nano35

Oh got damn... I see, the mks_robin_nano35_stm32 is for the V1.2, so I was using the wrong environment all the time?

ellensp, 2021-03-30:
Both are valid. But I am suspecting an issue in mks_robin_nano35, could you try mks_robin_nano35_stm32 ?

dskw, 2021-03-30:
Compiling worked fine so far. I'll try with that firmware later today or tomorrow and will update the ticket as soon as I have results!

thisiskeithb, 2021-03-30:
I’m wondering if this is related to Maple tone timer changes: https://github.com/MarlinFirmware/Marlin/issues/21429#issuecomment-809798444

ellensp, 2021-03-30:
I think it is quite likely.  related to that timer change.

thisiskeithb, 2021-03-30:
Tagging @rhapsodyv since he’s looking into it for another issue.

dskw, 2021-03-30:
I did two dry runs now (no filament loaded) and both times were executed correctly. So far I also did not see any tool switches during idle. I'll do some more testing and first test prints, but for now I'd say switching from mks_robin_nano35 to mks_robin_nano35_stm32 solved the issues.

dskw, 2021-03-30:
Alright the issue is officially resolved! Only intended tool switches happen now. Thanks for pointing towards the wrong board config!

thisiskeithb, 2021-03-30:
> Alright the issue is officially resolved! Only intended tool switches happen now. Thanks for pointing towards the wrong board config!

You weren't using the incorrect board config (PIO environment), but the one you were using has a bug that is causing issues with multiple boards.

rhapsodyv, 2021-03-30:
There’re two envs for nano v1 and v2 board. One uses the old framework maple. And the new env, uses a new framework stm32duino.

maple (mks_robin_nano35) will be discontinued soon. The new one (mks_robin_nano35_stm32) will be the default.

If your issue is solved when using the new env, that is great and you can keep using it. And you can close this.

Our objective is avoid fixing issues on old env, because it will be discontinued.

dskw, 2021-04-01:
As mentioned by rhapsodyv:
Issue was resolved by switching form the old mks_robin_nano35 environment to mks_robin_nano35_stm32.

closed 2021-04-01 by dskw