MarlinFirmware/Marlin #21473: [FR] (feature summary)

Grippy98, 2021-03-28:
### Is your feature request related to a problem? Please describe.

Some Chinese processor boards have switched to the GD32F103 which is an STM32 clone that is pin compatible and almost identical, but requires firmware changes for a faster clock and currently I am unable to compile successfully for this version of board.

If Creality chooses to go to the GD32 (or other manufacturers), this will be a problem in the future.

### Are you looking for hardware support?

GD32F103 Processor Support for Voxelab Aquilla (Ender 3 V2 Clone) 

### Describe the feature you want

Support for GD32F103 based motherboards.

TheDiscript, 2021-03-29:
For info, i have been having the same issue compiling for the GD32F103, specifically the RCT6 version in the same situation as the OP.  I have found the solution, the issue is caused by the memory difference from the original STM32F103RET6 that was on these boards.  The GD32F103RCT6 has 256kB flash and 48kB RAM vs the STM32F103RET6's 512kB flash and 64kB RAM.

To compile for the GD32F103RCT6 you need to set the ldscript memory as such on the creality.ld set up:
`MEMORY
{
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K - 40
  rom (rx)  : ORIGIN = 0x08007000, LENGTH = 256K - 28K
}`
Have been able to successfully compile and flash with this.

VinnyCordeiro, 2021-08-10:
Bigtreetech has been shipping SKR mini E3 boards using the GD32F103 microcontroller, the change was made without prior notice. https://twitter.com/3dpNero/status/1425121771406204932

eightheads, 2021-08-10:
> Bigtreetech has been shipping SKR mini E3 boards using the GD32F103 microcontroller, the change was made without prior notice. https://twitter.com/3dpNero/status/1425121771406204932

Nero doesn't know what he's talking about. This has been gone over already. No point in spreading false information once again.

VinnyCordeiro, 2021-08-10:
> Nero doesn't know what he's talking about. This has been gone over already. No point in spreading false information once again.

Not when a person buys 3 boards, all of them have the GD32 microcontroller, ask for replacement, and gets more boards with GD32 microcontroller. Call that whatever you want, it's in the wild and reported by multiple users.

Granted, BTT have released a recall on these boards, because that's a PR nightmare to release a product change without prior notice in such a specialized niche market. That doesn't change the fact that the change was made, is in the wild, and soon many users will start complaining about Marlin not working on their boards.

But, given your defensive attitude of blaming the messenger instead of trying to know what's happening, I should have stayed shut and just wait for the fireworks instead of trying to help. 'No good deeds goes unpunished', indeed. Have a nice day.

Grippy98, 2021-08-11:
I'm very confused by his statement of "Not fully compatible with Klipper". How? I've been running my Ender 3 with Klipper for MONTHS just fine.

eightheads, 2021-08-11:
> For the sake of completeness, Bigtreetech issued an official statement: https://twitter.com/BigTreeTech/status/1425478829867294722

Thank you for posting the official statement

Sebazzz, 2022-01-29:
Possibly helpful: https://github.com/platformio/platformio-core/issues/3927#issuecomment-888465652

thinkyhead, 2022-04-21:
I believe that Creality3D has posted a GD version of their firmware source code, but at the moment I don't have the link. I'll try to dig it up from my Discord history.

CrandellWS, 2022-06-19:
uh so... where are people at with this?

jdmz1, 2022-07-04:
Does anyone know where I can download the bugfix-2.0.9.3 release? I want to use H5labs's config files from #23806 but need to configure a couple things.

thisiskeithb, 2022-07-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

GiroroG66, 2022-08-08:
> For info, i have been having the same issue compiling for the GD32F103, specifically the RCT6 version in the same situation as the OP. I have found the solution, the issue is caused by the memory [...]

As far as I can tell, the clock frequency difference between GD32 and STM32 is also significant and causing some timing issues. The difference in cpu frequency is 120MHz vs 72MHz. This may be why there is a lot of salmon skin artifacts on STM builds run on GD boards, when that effect is not noticeable in creality GD board V4.2.2 stock 2.0.8.2 FW
The difference in cpu frequency is 120MHz vs 72MHz. F_CPU needs to be set to 120000000 and and BOARD_RCC_PLLMUL should be defined to RCC_PLL_MUL15 in boards_setup.cpp, but I don't know what all is needed to get that working the 'right' way. Even just setting F_CPU to 128,000,000 should be closer than 72MHz, and it seems like some work is already in place to make that frequency an option. 
 I don't want to kludge anything together until I find a way to revert back to stock Creality firmware, which nobody seems to have a copy of. At least, I can't find a non-BL touch version.  It's very frustrating that so many companies like Creality are allowed to sell closed forks of Marlin.

amweiss, 2022-08-08:
It looks like Creality posted 2 new firmware builds, including one specifically calling out GD chips if someone has a good way to reverse engineer the values out: https://www.crealitycloud.com/software-firmware/firmware?type=2&keyword=&system=0&subType=60067926229435d86f60789a&hardware=0

EvilGremlin, 2022-08-08:
Binaries are useless, PLL info we can get much easier from GD datasheet. GD 303 chip works fine at it is now, even if not up too full potentioal.
It'll be other story if creality publish sources that don't rely on ST libs, now that'll be useful.

Sebazzz, 2022-08-08:
Basically Creality is violating the ST license currently, right?

Met vriendelijke groet,
Sebastiaan Dammann

EvilGremlin, 2022-08-08:
Not really, ST lib license only state USB part explicitly. But China wouldn't care anyway i think, unless ST will bribe customs.

maxgerhardt, 2022-08-12:
Maintainer of [platform-gd32](https://github.com/CommunityGD32Cores/platform-gd32) and [ArduinoCore-GD32](https://github.com/CommunityGD32Cores/ArduinoCore-GD32) here -- how can we help the efforts? What's the current state? Any people actively working on it? 

>I believe that Creality3D has posted a GD version of their firmware source code, but at the moment I don't have the link. I'll try to dig it up from my Discord history.

Yeah they forked our core at https://github.com/vovan888/ArduinoCore-GD32/commits/main and there's some commits referencing a "BIGTREE_GD_TFT35_E3_V3_0", no idea what that is though... TFT and Ender 3 with GD32?

Their offiicial touch screen repo [also uses our platform](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/7c816bec99f181868b66ec679b9eefcd2d2130af/platformio.ini#L158-L165), but with the SPL framework instead of Arduino.

How can I get a printer that **for sure** has a GD32 chippy on it?

What exact functionality do the Marlin devs need from the Arduino core? Note that we already have all the general stuff, like GPIO, Serial, I²C, SPI, PWM, interrupts and even included Gigadevice's BSD 3-clause licensed USB stack (in the "Pluggable USB" style) for a USB CDC on the GD32F30x (e.g. GD32F303CC) series -- that means 0 copyright issues with ST Microelectronics. Plus the code of course makes full use of the maximum frequency of the GD32 chips and doesn't treat them like STM32 chips.

Does it make sense to start adding GD32 support right now with these features implemented?

EvilGremlin, 2022-08-12:
F103/303 and F407
As of now, only 303 chip works perfectly, so priority need is to patch up 103 and add 407
USB parts aren't really urgent as most (all?) the affected boards still use CH340

maxgerhardt, 2022-08-12:
>As of now, only 303 chip works perfectly, 

And with what config is that? Do people use the Maple core with a STM32 chip to compile for a GD32F303 that just happens to work?

GiroroG66, 2022-08-12:
> F103/303 and F407 As of now, only 303 chip works perfectly, so priority need is to patch up 103 and add 407 USB parts aren't really urgent as most (all?) the affected boards still use CH340

How do we know the GD 303 chip is working perfectly? If it's being clocked down to 72MHz then maybe. But if it's running at 120MHz, there would a clock mismatch that is likely causing problems with peripherals. That's what I was trying to say, before. I've heard of issues with surface artifacts/micro stepping and LCD incompatibility, but I can't try confirming anything until later this weekend. 

> How can I get a printer that **for sure** has a GD32 chippy on it?
A GD32F303RET6 is being used in the Creality 4.2.2.2 board currently being shipped inside the Ender 3 Pro B. I got mine at Microcenter and most shipping E3 pros are probably this model, but I have not seen any store pages updated. I can't find a "Ender 3 Pro B" version in any of Creality's official sites, but that's what was on the tiny sticker on the box. 
You can identify this model because it's power supply is not from MeanWell. It is a lookalike with a larger fan vent. 
I think Creality used the faster 303 (Cortex M4) over the 103 (M3) because the GD32F103 has significantly less memory than a ST32F103.

For the GD32F103 itself, I believe this this the chip used in the AnyCubic Vyper. I don't have a Vyper board to confirm. 
[https://www.anycubic.com/collections/for-vyper/products/motherboard-for-vyper](url)
Anycubic's newer Kobra Max (which I do have) uses HC32F460.

amweiss, 2022-08-12:
My Ender 3v2 from May (not the newest Neo version) has a 303. I can't say for sure if all of them do, but from other posts it seems like the recent builds of Ender 3v2 all have it.

EvilGremlin, 2022-08-12:
i mean build with standard stm32f103re_creality env works just fine, reportedly, no extra bugs
yep forgot HC460 chip, but this one is not by GD, not sure if libs will apply as is

maxgerhardt, 2022-08-13:
>i mean build with standard stm32f103re_creality env works just fine, reportedly, no extra bugs

Yeah and since that is using STM32Duino with STMicro's USB stack, everyone running that on GD32 chips is doing a license violation :/ I see room for improvement there too.

EvilGremlin, 2022-08-13:
They don't really - because boards don't have native USB. But end user wouldn't care anyway about STs arbitrary whims. So it's exclusively developers problem.

maxgerhardt, 2022-08-13:
You're right, although USB is by-standard enabled in `stm32_common`, the linker cleanly throws out all USB code in the final linker stage, so nothing of them is left in it. At least with the latest `Creality\Ender-3 Pro\CrealityV422` config I just tested. That's good to know.

maxgerhardt, 2022-08-23:
If you are using the `STM32F103RE_creality` [environment](https://github.com/MarlinFirmware/Marlin/blob/2a1c2e26ed38e55c52d569807884585a9523389c/ini/stm32f1.ini#L117-L152), that is using the `MARLIN_F103Rx` board, and its clock init code would be in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.cpp#L114-L130. Of course, the extra bits in the PLL register aren't available in the STM32 API, so you would have to do direct register writes to the RCU.

GiroroG66, 2022-08-23:
> If you are using the `STM32F103RE_creality` [environment](https://github.com/MarlinFirmware/Marlin/blob/2a1c2e26ed38e55c52d569807884585a9523389c/ini/stm32f1.ini#L117-L152), that is using the `MARLIN_F103Rx` board, and its clock init code would be in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.cpp#L114-L130. Of course, the extra bits in the PLL register aren't available in the STM32 API, so you would have to do direct register writes to the RCU. 120MHz from HSE + PLL is [for reference here](https://github.com/CommunityGD32Cores/gd32-pio-spl-package/blob/main/gd32/cmsis/variants/gd32f30x/system_gd32f30x.c#L810-L890).
> 
> But that is only half the rent -- the othe side is making the code recognize it now is running on e.g. 120MHz. Deep in the STM32 HAL there is a function ([here](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/STM32F1xx/system_stm32f1xx.c#L256-L362)) that retrieves the current system speed in Hz, which other parts of the code use to calculate the prescalers for timers or baudrate generators etc. Of course it has no idea about additional GD32 PLL capabilities, so it needs to be modified to recognize the correct speed of the MCU. Otherwise things like UART and PWM timers will be _very_ off.

Interesting. I was pretty far off. I was focused on assuming the clock was 120MHz, and trying to make the code able to handle that. It looks like if I'm understanding correctly, when the clock is configured correctly to 120Mhz, then the regs will be read out later and SystemCoreClock would be set to 120000000?

On the plus side, it looks like the GD32 board has the same 8MHz crystal, so if the clock is being set up the same way (they changed the register names in the datasheet, but maybe?) then it should be running at the same speed as the STM32, which explains why anything works. The GD datasheet that I have doesn't tell me the locations of these equivalent registers, so I haven't yet checked if the STM code is really configuring the GD chip to run at 72MHz. But if USB is working, then it's probably getting set up the same. 

By my reading of the GD datasheet, to get the clock running at 120MHz we would need
PLLPRESEL = 0
PREDV0 = 0
PLLSEL = /1
PLLMF = 15
SCS[1:0] = 0b10
But then I think we get into a mess with figuring out good values for AHB/APB, because 120 doesn't divide into 72/36. It would be easier on the peripherals to just get the GD32 running at 108MHz, but then we would probably also need to switch the USB clock over to the internal 48MHz oscillator. 

Or I could be completely wrong about all of this. I haven't worked with a HAL before.

Daasu72, 2022-09-09:
I was able to flash my GD32 303 board with STM32F103RE_creality environment.  I needed to make a few adjustments to my config file and now it will not re-flash.  I've tried changing the bin file name, but still wont re-flash.  Is that GD32 related?

chaimleib, 2022-09-10:
@Daasu72 I observed the same no-flash behavior, until I built a binary specifically for the GD platform.

phylyprobert, 2022-09-19:
@chaimleib Can you please share the binary? Thank you very much.

chaimleib, 2022-09-19:
@phylyprobert I had to use TH3D's project:
1. Download the code release for your printer at https://github.com/th3dstudio/UnifiedFirmware/releases
2. Uncomment the board version in Configuration.h
3. In Visual Studio Code with Auto Build Marlin extension, open the ABM panel.
4. Scroll down to the correct processor version and click the build button. (Mine was GD32F303RET6_creality_maple.)
5. Copy the firmware from .pio/build/<processor_version>/firmware-*.bin to your microSD.

I have not seen such an option to compile for GD processors in the mainline Marlin firmware yet.

Dasky14, 2022-09-19:
I wanted to write here that I also have a GD32F303, and I've noticed a lot of issues flashing firmware. I have one specific jyers marlin bin that works, but anything I compile manually using guides online or find from others just refuses to flash correctly.
I've been wrestling with this same chip trying to install klipper too, and finally just gave up.

maxgerhardt, 2022-09-19:
In [platform-gd32](https://github.com/CommunityGD32Cores/platform-gd32), we just use the [stm32f1x OpenOCD target](https://github.com/CommunityGD32Cores/platform-gd32/blob/main/boards/genericGD32F303CC.json#L25) to flash the firmwares. Or did you mean that upload is successful, but the firmware won't work afterwards?

phylyprobert, 2022-09-19:
@chaimleib Thank you very much friend!

GiroroG66, 2022-09-21:
When I was flashing between different firmware on my E3 GD board, I noticed I couldn't flash the same version of Marlin twice in a row. I don't just mean that I couldn't use the same file or the same file name. If I want to flash from 2.1.1 to a different build of 2.1.1 with different options, then I have to flash back to an older version in between (I used one of the Creality provided .bins). Otherwise, it would be like nothing happened. 
It's been a minute but I was having some other problems flashing too. I think one problems was that some of the "GD" files from crealty were configured for the fancy E3V2 LCD. Also, I think the board can't support SD cards that are bigger than, maybe 8GB. 

This might be why some people are having trouble flashing

Matombo, 2022-10-20:
Anyone has a working config for the Ender 3 V2 with CR Touch?

The creality gd firmware with bl touch support will not boot for me -.-

Got the 4.2.2 board with GD32F303 RET6

hassansaleh31, 2022-11-06:
I downloaded the latest version of Marlin (2.1.1), Enabled Mesh Bed Leveling for manual mesh leveling, uncommented `#define RESTORE_LEVELING_AFTER_G28` and compiled with STM32F103RE_creality env.

The printer works fine even though I lost my previously saved setting, but I went ahead and finished the manual bed Leveling procedure, stored the configurations, and added `M240 S1 ; Restore Manual Mesh` after `G28`.

The thing is that the Bed Z setting on the LCD is not changing anything no matter how high or low I set it to. I also noticed that the mesh was not applied (fillament is a lot thinner in the middle of the bed)

Machine: Ender 3 V1 with 4.2.2 board and GD32F303.
Start G-Code:
```
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
M240 S1 ; Restore Manual Mesh
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```