MarlinFirmware/Marlin #21474: [BUG]

Deneteus, 2021-03-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Cannot compile Wanhao D6 example profile without errors.

I had been wanting to create a new stock Wanhao D6 profile for 2021 and then add the changes for my Bondtech extruder & Microswiss hot end as well as enable some of the newer features such as LA 1.5 and the A4988 optimization settings that didn't exist in the original stock firmware. I had made one in 2019 but I lost the config files and had to start from scratch.

1. I used Github Desktop and downloaded Gary Chen's original D6 configuration files. 
https://github.com/garychen99/Duplicator-6-firmware-version-3.1

2. Downloaded the current version of bigfix-2.0.x code on 3/26/21.

3. Opened Beyond Compare and started copying over all the D6 specific defaults to the Bugfix configuration.h & configuration_adv files.

4. Added all the config changes I wanted and then loaded it up in VSCODE.

5. Nothing but errors. The first thing that I got an error about was that ATMEGA2560 couldn't be used and I got errors about the LCD. The Wanhao D6 uses the SSD1306 with a couple specific settings. I was able to resolve some of the errors but I was still left with some code issues having to do with the LCD. 

6. To be sure I wasn't crazy I made sure the configuration dates hadn't changed in github and I downloaded the D6 profile from configs and I still couldn't compile without the errors posted below. 

### Bug Timeline

New. Since the changes to the variable names.

### Expected behavior

I expected it to compile without errors.

### Actual behavior

Compiling .pio\build\mega2560ext\src\src\lcd\fontutils.cpp.o
In file included from Marlin\src\lcd\dogm\../menu/../marlinui.h:28:0,
                 from Marlin\src\lcd\dogm\../menu/menu.h:24,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../buttons.h:164:0: warning: "BUTTON_PRESSED" redefined
 #define BUTTON_PRESSED(BN) (_BUTTON_PRESSED_##BN)
 
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:150:0: note: this is the location of the previous definition
   #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
Compiling .pio\build\mega2560ext\src\src\lcd\lcdprint.cpp.o
 
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
Compiling .pio\build\mega2560ext\src\src\lcd\marlinui.cpp.o
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h:75:0: warning: "LCD_UPDATE_INTERVAL" redefined
   #define LCD_UPDATE_INTERVAL TERN(HAS_TOUCH_BUTTONS, 50, 100)
 
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:77:0: note: this is the location of the previous definition
   #define LCD_UPDATE_INTERVAL TERN(HAS_TOUCH_XPT2046, 50, 100)
 
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h:63:8: error: multiple definition of 'enum LCDViewAction'
   enum LCDViewAction : uint8_t {
        ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:65:8: note: previous definition here
   enum LCDViewAction : uint8_t {
        ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h:82:134: warning: default argument given for parameter 5 of 'void _wrap_string(uint8_t&, uint8_t&, const char*, read_byte_cb_t, bool)' [-fpermissive]
       void _wrap_string(uint8_t &col, uint8_t &row, const char * const string, read_byte_cb_t cb_read_byte, const bool wordwrap=false);
                                                                                                                                      ^
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:83:10: note: previous specification in 'void _wrap_string(uint8_t&, uint8_t&, const char*, read_byte_cb_t, bool)' here
     void _wrap_string(uint8_t &col, uint8_t &row, const char * const string, read_byte_cb_t cb_read_byte, const bool wordwrap=false);
          ^~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h: In function 'void wrap_string_P(uint8_t&, uint8_t&, const char*, bool)':
Marlin\src\lcd\dogm\../menu/../marlinui.h:83:19: error: redefinition of 'void wrap_string_P(uint8_t&, uint8_t&, const char*, bool)'
       inline void wrap_string_P(uint8_t &col, uint8_t &row, PGM_P const pstr, const bool wordwrap=false) { _wrap_string(col, row, pstr, read_byte_rom, wordwrap); }
                   ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:84:17: note: 'void wrap_string_P(uint8_t&, uint8_t&, const char*, bool)' previously defined here
     inline void wrap_string_P(uint8_t &col, uint8_t &row, PGM_P const pstr, const bool wordwrap=false) { _wrap_string(col, row, pstr, read_byte_rom, wordwrap); }
                 ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h: In function 'void wrap_string(uint8_t&, uint8_t&, const char*, bool)':
Marlin\src\lcd\dogm\../menu/../marlinui.h:84:19: error: redefinition of 'void wrap_string(uint8_t&, uint8_t&, const char*, bool)'
       inline void wrap_string(uint8_t &col, uint8_t &row, const char * const string, const bool wordwrap=false) { _wrap_string(col, row, string, read_byte_ram, wordwrap); }
                   ^~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:85:17: note: 'void wrap_string(uint8_t&, uint8_t&, const char*, bool)' previously defined here
     inline void wrap_string(uint8_t &col, uint8_t &row, const char * const string, const bool wordwrap=false) { _wrap_string(col, row, string, read_byte_ram, wordwrap); }
                 ^~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h: At global scope:
Marlin\src\lcd\dogm\../menu/../marlinui.h:95:8: error: multiple definition of 'enum MarlinFont'
   enum MarlinFont : uint8_t {
        ^~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:235:8: note: previous definition here
   enum MarlinFont : uint8_t {
        ^~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
Marlin\src\lcd\dogm\../menu/../marlinui.h:119:5: error: conflicting declaration 'typedef struct preheat_t preheat_t'
   } preheat_t;
     ^~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:253:5: note: previous declaration as 'typedef struct preheat_t preheat_t'
   } preheat_t;
     ^~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mega2560ext\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
============================================================================== [FAILED] Took 88.88 seconds ==============================================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560ext    FAILED    00:01:28.878
========================================================================= 1 failed, 0 succeeded in 00:01:28.878 =========================================================================
The terminal process "C:\Users\Derrick Smith\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Steps to Reproduce

1. Goto the Configurations and copy the Wanhao D6 profile over the stock configs downloaded from the latest Bugfix release.
2. Open VSCODE and open the project and Build it.

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 020008

### Printer model

Wanhao/Duplicator 6

thisiskeithb, 2021-03-29:
> I used Github Desktop and downloaded Gary Chen's original D6 configuration files.
https://github.com/garychen99/Duplicator-6-firmware-version-3.1

We cannot support forks here.

closed 2021-03-29 by thisiskeithb

Deneteus, 2021-03-29:
@thisiskeithb This is not a port. You didn't even read the issue thoroughly. What I was trying to do was copy over the config settings to the current Marlin bugfix configuration files and then compile and when that didn't work I tested the Bugfix with the Example config and it failed as well so I know it wasn't my fault because the Example config doesn't work.

ellensp, 2021-03-29:
Config url you provided. https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Wanhao/Duplicator%206

Readme for the Configurations.
```
NB If you are downloading these configuration files in a zip file from the CODE button, you first must select the branch that matches the version of Marlin you are compiling.

Branches
Release: For most Marlin users. Select the version that matches the Marlin version you are compiling.

Bugfix: For users compiling the bugfix branch of Marlin.

Import: For developers maintaining the Configuration examples.

```
for bugfix code you need to use bugfix branch
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Wanhao/Duplicator%206

This compiles without out any issues.

Deneteus, 2021-03-29:
I did and it did not compile without issues.

Deneteus, 2021-03-29:
Steps to Reproduce
Goto the Configurations and copy the Wanhao D6 example profile over the stock configs downloaded from the latest Bugfix release. https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Wanhao/Duplicator%206
Open VSCODE and open the project and Build it.
Look at the errors.

Deneteus, 2021-03-29:
Why even ask for documentation if you aren't going to read it.

I did not say I used that branch to build it. I said I copied over the configuration settings to the current branch.

Deneteus, 2021-03-29:
What version does this say?
[image]

ellensp, 2021-03-29:
stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch! stop using import branch!

Deneteus, 2021-03-29:
I wasn't using import branch I just copied the wrong link.

ellensp, 2021-03-29:
They your doing something wrong.  
latest bugfix + https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Wanhao/Duplicator%206
no modifications.

compiles without issues.

My guess is you extracted bugfix code over marlin release code And you now have a mix of both lots of code 
Since many files changes names. resulting in many redefinition errors.

Delete you copy of marlin. Start again

Deneteus, 2021-03-29:
Just for you I wiped the repository and re-downloaded it.
[image]

Copied over these configs.

[image]

Into the current version of Marlin Bugfix.

[image]

Gives this error.

[image]

Manual update the Platformio.ini with the correct board.
[image]

Fails to compile when it reaches the LCD section.

[image]

ellensp, 2021-03-29:
from your first image, still import branch

[image]

Deneteus, 2021-03-29:
Yeah I caught that after I posted it. I am re-wiping out everything again. I had already done this before like 6 times.

[image]

Deneteus, 2021-03-29:
[image]

Deneteus, 2021-03-29:
When you wipe it out and reload it the window defaults to import branch,

[image]

thisiskeithb, 2021-03-29:
For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

Deneteus, 2021-03-29:
Now it's the correct branch in the dropdown. 

[image]

[image]

[image]

@thisiskeithb Could you stop being patronizing for one second. I am in those groups and the forum. 

[image]

[image]

This is what happens.

[image]

ellensp, 2021-03-29:
so now you have bugfix config files and marlin 2.0.x code... still miss matched....

Deneteus, 2021-03-29:
I looked at the window the other workspace was still open. Just closed it and re-ran it.

[image]

I am going to manually download the files instead of using github and retest it.

thisiskeithb, 2021-03-29:
> I am in those groups and the forum.

Please use them. This is not what the issue queue is for.

Deneteus, 2021-03-29:
[image]

> Executing task in folder Marlin: C:\Users\Derrick Smith\.platformio\penv\Scripts\platformio.exe run <

Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.2.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\babystep.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\mega2560\src\src\feature\caselight.cpp.o
Compiling .pio\build\mega2560\src\src\feature\leds\leds.cpp.o
Compiling .pio\build\mega2560\src\src\feature\leds\pca9632.cpp.o
Compiling .pio\build\mega2560\src\src\feature\leds\printer_event_leds.cpp.o
Compiling .pio\build\mega2560\src\src\feature\pause.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M211.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M280.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M350_M351.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M400.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M80_M81.cpp.o
[... 165 lines cut ...]
Marlin\src\module\../inc/../core/macros.h:196:29: note: in expansion of macro 'THIRD'
 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.
                             ^~~~~
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^~~~~
Marlin\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                            ^~~~
*** [.pio\build\mega2560\src\src\module\motion.cpp.o] Error 1
============================================================================== [FAILED] Took 94.32 seconds ==============================================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:01:34.322
========================================================================= 1 failed, 0 succeeded in 00:01:34.322 =========================================================================
The terminal process "C:\Users\Derrick Smith\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ellensp, 2021-03-29:
you have selected 2.0.x branch of marlin code
[image]

not bugfix.

You MUST match your version of config with the code

rhapsodyv, 2021-03-29:
"ultralcd.h" doesn't exist anymore. It's a build issue. You have old and new files mixed in the same folder.

You should ask for help on one of Malin groups/forums:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server