MarlinFirmware/Marlin #21475: [BUG] Build failure if AUTO_POWER_CONTROL is enabled but POWER_TIMEOUT is disabled

Pyro-Fox, 2021-03-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

under the Power Supply Control section, if I enable PSU_CONTROL and AUTO_POWER_CONTROL, but disable POWER_TIMEOUT, I get the following build failure from PIO:

```
                 from Marlin\src\feature\../inc/MarlinConfig.h:54,
                 from Marlin\src\feature\power.cpp:27:
Marlin\src\feature\power.cpp: In static member function 'static void Power::check()':
Marlin\src\feature\power.cpp:109:66: error: 'POWER_TIMEOUT' was not declared in this scope; did you mean 'ERR_TIMOUT'?
  109 |     else if (!lastPowerOn || ELAPSED(ms, lastPowerOn + SEC_TO_MS(POWER_TIMEOUT)))
      |                                                                  ^~~~~~~~~~~~~
c:\marlin_git\marlin\marlin\src\core\millis_t.h:32:43: note: in definition of macro 'PENDING'
   32 | #define PENDING(NOW,SOON) ((int32_t)(NOW-(SOON))<0)
      |                                           ^~~~
Marlin\src\feature\power.cpp:109:30: note: in expansion of macro 'ELAPSED'
  109 |     else if (!lastPowerOn || ELAPSED(ms, lastPowerOn + SEC_TO_MS(POWER_TIMEOUT)))
      |                              ^~~~~~~
Marlin\src\feature\power.cpp:109:56: note: in expansion of macro 'SEC_TO_MS'
  109 |     else if (!lastPowerOn || ELAPSED(ms, lastPowerOn + SEC_TO_MS(POWER_TIMEOUT)))
      |                                                        ^~~~~~~~~
*** [.pio\build\LPC1768\src\src\feature\power.cpp.o] Error 1
```

### Bug Timeline

New

### Expected behavior

I expect it to build with timeout disabled

### Actual behavior

It doesn't build

### Version of Marlin Firmware

2.0.x-bugfix (Distribution date 2021-03-29)

### Printer model

Custom

### Electronics

BTT SKR 1.3

[Marlin SKR 1.3 files.zip](https://github.com/MarlinFirmware/Marlin/files/6219115/Marlin.SKR.1.3.files.zip)

[PIO Build Failure.log](https://github.com/MarlinFirmware/Marlin/files/6219118/PIO.Build.Failure.log)

ams2222891, 2021-03-29:
> ### Did you test the latest `bugfix-2.0.x` code?
>
> Yes, and the problem still exists.
>
> ### Bug Description
>
> under the Power Supply Control section, if I enable PSU_CONTROL and AUTO_POWER_CONTROL, but [...]

thisiskeithb, 2021-05-29:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/21771

closed 2021-05-29 by thisiskeithb