MarlinFirmware/Marlin #21479: [FR]Animated full screen splash screen

thelonious35, 2021-03-29:
### Describe the feature you want

Hello. I want to put my own animation on the splash in full screen and disable the marlin logo and url

thisiskeithb, 2021-03-29:
Enable `BOOT_MARLIN_LOGO_ANIMATED`.

[Here's a `_bootscreen.h` example](https://github.com/MarlinFirmware/Configurations/blob/75b4ef126409da4117381c67fc138c138c345132/config/examples/AnimationExample/_Bootscreen.h) you can use to get started on your own design.

closed 2021-03-29 by thisiskeithb

thelonious35, 2021-03-29:
> Etkinleştirin `BOOT_MARLIN_LOGO_ANIMATED`.
> 
> [İşte`_Bootscreen.h`](https://github.com/MarlinFirmware/Configurations/blob/75b4ef126409da4117381c67fc138c138c345132/config/examples/AnimationExample/_Bootscreen.h) kendi tasarımınıza başlamak için kullanabileceğiniz [animasyonlu bir ](https://github.com/MarlinFirmware/Configurations/blob/75b4ef126409da4117381c67fc138c138c345132/config/examples/AnimationExample/_Bootscreen.h)[örnek](https://github.com/MarlinFirmware/Configurations/blob/75b4ef126409da4117381c67fc138c138c345132/config/examples/AnimationExample/_Bootscreen.h) .

ı opened BOOT_MARLIN_LOGO_ANIMATED and changed the codes you gave but it gives an error

thisiskeithb, 2021-03-29:
For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

thelonious35, 2021-03-29:
> Yapılandırma ve sorun giderme konusunda en iyi sonuçları almak için lütfen aşağıdaki kaynakları kullanın:
> 
> * [Marlin Dokümantasyonu](https://marlinfw.org) MarlinFW.org'da barındırılıyor
> * [Marlin Forum](https://forums.reprap.org/list.php?415) , RepRap.org'da barındırılıyor
> * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Grubu
> * [3B Yazıcılar için Marlin Firmware](https://www.facebook.com/groups/3Dtechtalk/) Facebook Grubu
> * [](https://www.youtube.com/results?search_query=marlin+configuration)YouTube'da [Marlin Yapılandırması](https://www.youtube.com/results?search_query=marlin+configuration)
> * [Marlin Firmware](https://discord.gg/n5NJ59y) Discord sunucusu

I couldn't find it anywhere, I guess I have to give up anyway thanks.