MarlinFirmware/Marlin #21482: [FR] Suggestion to change typedefs to their fast counterparts

D337z, 2021-03-30:
### Is your feature request related to a problem? Please describe.

Marlin is designed to work with many different processors including 8, 16, and 32 bit using much of the same code. But it uses the same typedefs sizes for each.

### Describe the feature you want

By using fast types such as int_fast32 and the like, the size of types will be optimized for each processor based on its capabilities. If a 32-bit integer is faster, it will be a 32-bit. If an 8-bit is faster and it will fit, it will be an 8-bit.
Code can then be reduced in size and increased in execution rate.

ManuelMcLure, 2021-03-30:
As far as I can tell, the only 8 bit processors supported by Marlin are AVR, and on those there are no native instructions to access 16 or 32 bit values in memory - instead you need two load/store instructions to load/store a 16 bit value and 4 load/store instructions to load/store a 32 bit value. So I doubt that using 16 or 32 bit variables will ever be a performance benefit on AVR.

D337z, 2021-03-30:
Currently, yes. But, suppose you had a variable that you have set for
32-bit that would possibly fit in 8 or 16 bits. Regardless of the actual
size, you're using 32 bits which would require those 4 and 2 load/stores
respectively. But, if you used fast_int8_t, you could reduce it to 1 while
still having the ability to increase it as necessary up to 32. It's a start
size rather than absolute.
This would make it able to be handled in a single load/store for small
types while not forcing it to become a large type unnecessarily.
And yeah, I forgot that Atmel is 16-bit. But many of these numbers and
characters should be able to fit in a relatively small size. Code size,
alone, should see an improvement after compilation.
Anyway, it was just a suggestion and something you might play around with
to see what works if you're feeling froggy.
But I could see some potential benefits to using this in reducing
processing times and code size in the long-run.

Thanks.