MarlinFirmware/Marlin #21486: anycubic_i3mega_lcd.cpp and compiling errors

francko76, 2021-03-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Visual Studio

In file included from Marlin\src\lcd\extui\lib\anycubic_i3mega\../../../../inc/../core/boards.h:24:0,
                 from Marlin\src\lcd\extui\lib\anycubic_i3mega\../../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:22:
Marlin\src\lcd\extui\lib\anycubic_i3mega\../../../../inc/../core/macros.h: In instantiation of 'constexpr void LIMIT(V&, N1, N2) [with V = const float; N1 = int; N2 = int]':
Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:587:13:   required from here
Marlin\src\lcd\extui\lib\anycubic_i3mega\../../../../inc/../core/macros.h:126:19: error: assignment of read-only reference 'v'
     if (n1 > v) v = n1;
                   ^
Marlin\src\lcd\extui\lib\anycubic_i3mega\../../../../inc/../core/macros.h:127:24: error: assignment of read-only reference 'v'
     else if (n2 < v) v = n2;
                        ^
Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp: In static member function 'static void AnycubicTFTClass::GetCommandFromTFT()':
Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:587:43: error: void value not ignored as it ought to be
             SEND_PGM_VAL("A4V ", int(LIMIT(fanPercent, 0, 100)));
                                           ^
Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:37:66: note: in definition of macro 'SEND_PGM_VAL'
 #define SEND_PGM_VAL(x,y) (send_P(PSTR(x)), sendLine(i16tostr3rj(y)))
                                                                  ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
Compiling .pio\build\mega2560\src\src\lcd\marlinui.cpp.o
*** [.pio\build\mega2560\src\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp.o] Error 1

Arduino

c:\tmp\sketch\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp: In static member function 'static void AnycubicTFTClass::GetCommandFromTFT()':
c:\tmp\sketch\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:587:43: error: void value not ignored as it ought to be
             SEND_PGM_VAL("A4V ", int(LIMIT(fanPercent, 0, 100)));
                                      ~~~~~^~~~~~~~~~~~~~~~~~~~
c:\tmp\sketch\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:37:66: note: in definition of macro 'SEND_PGM_VAL'
 #define SEND_PGM_VAL(x,y) (send_P(PSTR(x)), sendLine(i16tostr3rj(y)))
                                                                  ^
In file included from c:\tmp\sketch\src\core\boards.h:24:0,
                 from c:\tmp\sketch\src\inc\marlinconfigpre.h:37,
                 from c:\tmp\sketch\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:22:
c:\tmp\sketch\src\core\macros.h: In instantiation of 'constexpr void LIMIT(V&, N1, N2) [with V = const float; N1 = int; N2 = int]':
c:\tmp\sketch\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp:587:13:   required from here
c:\tmp\sketch\src\core\macros.h:126:19: error: assignment of read-only reference 'v'
     if (n1 > v) v = n1;
                 ~~^~~~
c:\tmp\sketch\src\core\macros.h:127:24: error: assignment of read-only reference 'v'
     else if (n2 < v) v = n2;
                      ~~^~~~
exit status 1

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Anycubic i3 mega s

ellensp, 2021-03-30:
Please attach your configuration files.

francko76, 2021-04-04:
I fixed temporarly for compiling changing this

anycubic_i3mega_lcd.cpp
row 587

[image]

closed 2021-04-13 by thinkyhead