MarlinFirmware/Marlin #21490: [FR] Transfering compressed Gcode and then uncompressing it on board

buzzcola3, 2021-03-31:
### Describe the feature you want

since transfering files through serial is quite slow, it would be nice to be able to send compressed gcode files (zip or something) and then decompress it on board. Not sure how possible that is, but at least on the 32bit boards it might work quite fast.
and since Gcode can be quite repetitive it has quite high compression ratio. compressed gcode is usually at least 3-4x smaller 

it would be nice if M29 automagically detected .zip extension, automagically decompressed it, md5'd it (if md5 is available and came with the zip), saved the gcode and deleted the source zip

ellensp, 2021-03-31:
both meatpack and binary transfer do this while sending over serial.

buzzcola3, 2021-03-31:
> both meatpack and binary transfer do this while sending over serial.

oh, thats awesome. so it already kinda exists

closed 2021-03-31 by buzzcola3