MarlinFirmware/Marlin #21493: [BUG] LCD Errors during build

tinbasher, 2021-03-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Attempting to build from BugFix 2.0.x.  Build fails concerning LCD options.

### Bug Timeline

New bug

### Expected behavior

To complete the build process.

### Actual behavior

Build fails.

### Steps to Reproduce

1. Configure .h files
2. Build

### Version of Marlin Firmware

Current is 2.0.7.2

### Printer model

Ender 3

### Electronics

BTT SKR Mini E3 V2.0

### Add-ons

TFT35 V3

### Host Software

OctoPrint

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6238109/Configuration.zip)

ellensp, 2021-04-01:
Provided configs compiles without any issues

ellensp, 2021-04-01:
no error provided...   cannot reproduce. Closing.

closed 2021-04-01 by ellensp