MarlinFirmware/Marlin #21504: [FR] ADC M105 - Report Temperatures SHOW_TEMP_ADC_VALUES

pimchest, 2021-04-01:
### Is your feature request related to a problem? Please describe.

I expect ADC values ​​to range from 0 - 1000, not 0 - 4000:

23:39:56.359 : T:46.25 /0.00 (3648.00) B:23.91 /0.00 (3915.00) @:0 B@:0
23:39:57.359 : T:46.19 /0.00 (3649.00) B:23.59 /0.00 (3917.00) @:0 B@:0
23:39:58.359 : T:46.13 /0.00 (3650.00) B:23.59 /0.00 (3917.00) @:0 B@:0
23:39:59.359 : T:45.94 /0.00 (3653.00) B:23.44 /0.00 (3918.00) @:0 B@:0
23:40:00.359 : T:45.94 /0.00 (3653.00) B:23.75 /0.00 (3916.00) @:0 B@:0
23:40:01.360 : T:45.94 /0.00 (3653.00) B:23.91 /0.00 (3915.00) @:0 B@:0
23:40:02.378 : T:45.81 /0.00 (3654.00) B:23.75 /0.00 (3916.00) @:0 B@:0
23:40:03.376 : T:45.75 /0.00 (3656.00) B:23.59 /0.00 (3917.00) @:0 B@:0

### Are you looking for hardware support?

Board: MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
Trianglelab ATC Semitec 104GT-2 104NT-4-R025H42G 
File: thermistor_5.h

### Describe the feature you want

I'm trying to create my config file with thermistor_502.h as an example, but I can't figure out if it's correct to divide my serial port values ​​by 4.

I want to create my table of ADC values ​​and corresponding temperatures, but ran into the problem that my ADC values ​​are very large. I found a similar config file, but all values ​​are reduced by 4.

I don't understand the original error. Incorrect return values ​​from the ADC, or incorrectly configured config file or something else? Perhaps my sensor is broken and therefore incorrect values ​​are returned from the ADC. At the moment, the values ​​on the printer and the real ones differ by about 15 degrees

thisiskeithb, 2021-04-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-01 by thisiskeithb

pimchest, 2021-04-02:
https://reprap.org/forum/read.php?1,872228,872228