MarlinFirmware/Marlin #21508: [BUG] FLSUN_Hispeed not Compatible with FLSU_Hispeed

Savage79, 2021-04-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Motherboard not recognized, Build environment 'flsun_hispeedv1' is incompatible with BOARD_FLSUN_HISPEED. Use one of these: flsun_hispeed

### Bug Timeline

Unknown

### Expected behavior

Expected board to be recognized

### Actual behavior

Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'flsun_hispeedv1' is incompatible with BOARD_FLSUN_HISPEED. Use one of these: flsun_hispeed

### Steps to Reproduce

Downloaded fresh copy of Marlin Bugfix April 1,2021
Retrieved config files from FLSUN_QQS_Pro 
Copy and past into Marlin folder
Change  default_envs = flsun_hispeedv1
save and compile

### Version of Marlin Firmware

Marlin Bugfix 2.0.X

### Printer model

Flsun QQS Pro

### Electronics

Flsun HISPEED V1 

### Add-ons

TMC2208 in UART

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

No need to post a zip of the files, they are AS IS from the link provided for config. examples, no changes to config files.

[FLSUN_Config.zip](https://github.com/MarlinFirmware/Marlin/files/6247314/FLSUN_Config.zip)

ellensp, 2021-04-02:
in pins.h is the list of valid environments

```
#elif MB(FLSUN_HISPEED)
  #include "stm32f1/pins_FLSUN_HISPEED.h"       // STM32F1                                env:flsun_hispeed

```
does not match `[env:flsun_hispeedv1]` in platfomio.ini

they either both need to be v1 or nether of them

ellensp, 2021-04-02:
https://github.com/MarlinFirmware/Marlin/pull/19793 added the following correctly.
```
#define BOARD_FLSUN_HISPEED 4039 // FLSUN HiSpeedV1 (STM32F103VET6)
#elif MB(FLSUN_HISPEED)
  #include "stm32f1/pins_FLSUN_HISPEED.h"       // STM32F1                                env:flsun_hispeed
[env:flsun_hispeed]
```
https://github.com/MarlinFirmware/Marlin/pull/20354 renamed [env:flsun_hispeed] to [env:flsun_hispeedv1] without updating pins.h

ellensp, 2021-04-02:
PR has been created to fix this

slowbro, 2021-07-03:
PR #21510 was merged, this can be closed.

closed 2021-07-03 by thisiskeithb