MarlinFirmware/Marlin #21509: [BUG] BTT SKR 1.4 3DTouch pin wont deploy

1liminal1, 2021-04-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

3D Touch does not deploy pin on G28 or any other probing or testing G commands on a fresh install of Malin 2.0.7.2 or the latest bugfix. 

I'm using the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and it will just not deploy and crash into the bed. 

I have tried using the dedicated prope pins on the SKR, but that errors the board. like is unplugged, ie it is "working" but the probe pin jst does not deploy. 

### Bug Timeline

It seems like others are having this issue. But I cant seem to get a definitive answer on a fix 

### Expected behavior

The pin deploys and I can home correctly

### Actual behavior

Pin will not deploy and nozzle crashes into bed

### Steps to Reproduce

G28
G29 P1

### Version of Marlin Firmware

2.0 bugix downloaded 02-04-2021

### Printer model

AM8

### Electronics

SKR 1.4 (non turbo)

### Add-ons

3D touch

### Your Slicer

Cura

### Host Software

OctoPrint

I can post if needed, but its pretty stock standed

#define USE_ZMIN_PLUG

#define BLTOUCH

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Nothing else has been enabled that is outside stock Marlin for the context of BLtouch 

Also, I don't want to use the probe pins, I have a JXT connection already set up for the Z min endstop and don't want to recable 

Thanks

thisiskeithb, 2021-04-02:
Please attach your configs.

Is there a version of Marlin that works?

I’m running a copy of bugfix that’s a couple days old on an SKR 1.4 with a genuine BLTouch (using the probe pin) that works fine.

1liminal1, 2021-04-02:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6247583/Configuration_adv.zip)

Hey thanks for that, It was working fine on my MKS Gen L, but since switching to the SKR 1.4 with 2.0.X its not. 

> I’m running a copy of bugfix that’s a couple days old on an SKR 1.4 with a genuine BLTouch (using the probe pin) that works fine.

This is a 3Dtouch

Do you have anything special set for pins ect or just the std; 

#define USE_ZMIN_PLUG

#define BLTOUCH

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Thanks

thisiskeithb, 2021-04-02:
I’d hop on one of the resources below for further troubleshooting:

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-02 by thisiskeithb