MarlinFirmware/Marlin #21517: Error in source, fails to compile.

bug12345678, 2021-04-02:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Error in marlin-2.0.x\Marlin\src\pins\pins.h 
line 747 #error "Unknown MOTHERBOARD value set in Configuration.h" 
on "#error".

### Bug Timeline

I don't know.

### Expected behavior

It compiles.

### Actual behavior

It doesn't compile.

### Steps to Reproduce

1. Open it in vs code.
2. Try to compile it.
3. Check out the problems.

### Version of Marlin Firmware

I don't know, actually, I downloaded it sometime in march.

### Printer model

Custom.

### Electronics

bigtreetech skr 1.3

### Add-ons

None that I am aware of.

I am not experienced at all with the programming and don't know how to fix it myself.

> Executing task: C:\Users\USER\.platformio\penv\Scripts\platformio.exe run --environment LPC1768 <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:37,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/boards.h:395:64: error: operator '==' has no right operand
  395 | #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
      |                                                                ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/macros.h:159:29: note: in expansion of macro '_MB_1'
  159 | #define _DO_1(W,C,A)       (_##W##_1(A))
      |                             ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/macros.h:171:28: note: in expansion of macro '_DO_1'
  171 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/macros.h:172:28: note: in expansion of macro '__DO_N'
  172 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/macros.h:173:28: note: in expansion of macro '_DO_N'
  173 | #define DO(W,C,V...)       _DO_N(W,C,NUM_ARGS(V),V)
      |                            ^~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/boards.h:396:19: note: in expansion of macro 'DO'
  396 | #define MB(V...)  DO(MB,||,V)
      |                   ^~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:395:7: note: in expansion of macro 'MB'
  395 | #elif MB(BTT_SKR_V1_3)
      |       ^~
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:747:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
  747 |     #error "Unknown MOTHERBOARD value set in Configuration.h"
      |      ^~~~~
CalledProcessError: Command 'arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\USER\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\USER\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\USER\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\USER\Downloads\Marlin-konfiguration\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 286:
    apply_features_config()
  File "C:\Users\USER\Downloads\Marlin-konfiguration\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\USER\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\USER\Downloads\Marlin-konfiguration\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 262:
    load_marlin_features()
  File "C:\Users\USER\Downloads\Marlin-konfiguration\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\USER\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\USER\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
================================================================================================================================================ [FAILED] Took 1.21 seconds ================================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:01.208
========================================================================================================================================== 1 failed, 0 succeeded in 00:00:01.208 ========================================================================================================================================== 
The terminal process "C:\Users\USER\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'LPC1768'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

[Marlin-konfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/6250841/Marlin-konfiguration.zip)

I have tried to modify it to work with my hardware, but that doesn't seem to be the issue, also i haven't touched pins.h much at all, if anything.

rubino25, 2021-04-02:
Same error! Can't compile

ellensp, 2021-04-03:
you replaced
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif

with #define BOARD_BTT_SKR_V1_3   

when it need to be at minimum #define MOTHERBOARD BOARD_BTT_SKR_V1_3

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-03 by ellensp