MarlinFirmware/Marlin #21520: [BUG] Head crashes after Home and Z-Align

kpishere, 2021-04-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First homing operation works, second one does not.  It crashes to the right in a second home operation.

### Bug Timeline

Has been there since December that I've noticed.

### Expected behavior

I'd expect the X-home operation to find home again by moving all the way to the left and use TMC2209 sensor-less homing feature.

### Actual behavior

After first homing operation, it appears to take the centre of the bed as 0 in the X coordinate and then move from there in the next homing operation.  NOTE: when one prints after only one home operation on boot, the coordinate system appears to be correct in that prints form in middle of bed as expected.

### Steps to Reproduce

After boot 1) Select from menu Motion > Auto Home, then 2) Motion > Auto Z-Align.  The head with crash in the far right.  It looks like zero coordinate system is starting from head being in mid-point of bed rather than using TMC2209 sensor-less end-stop to fine home.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 31 2021 00:07:27) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Itopie 400+ EXTRUDER_COUNT:1 UUID:41B1EDA1-B268-466E-8B01-C6458C23FFF2

### Printer model

iTopie400+

### Electronics

SMART RAMPS with MEGA2560 and TMC2209 drivers

### Add-ons

Fixed Induction probe

### Your Slicer

Cura

### Host Software

SD Card (headless)

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/6251317/Archive.zip)

Source is in GIT under iTopie400+ branch :  [github](https://github.com/kpishere/Marlin)

[Youtube-recreate-problem](https://youtu.be/MOQoHRY6-fw)

M115:  

FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 31 2021 00:07:27) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Itopie 400+ EXTRUDER_COUNT:1 UUID:41B1EDA1-B268-466E-8B01-C6458C23FFF2
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0

kpishere, 2021-04-06:
M111S32 output from first G28:  

>>> G28  X0.00 Y0.00 Z0.25
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X-44.00 Y2.00 Z-0.60 (Left-Back & Below Nozzle)
Auto Bed Leveling: LINEAR (disabled)

Raise Z (before homing) by 5.00
>>> do_blocking_move_to  X0.00 Y0.00 Z0.25
>  X0.00 Y0.00 Z5.00
<<< do_blocking_move_to  X0.00 Y0.00 Z5.00
>>> homeaxis(88)
Home Fast: -348.00mm
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(X, -348.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z5.00
Endstop 88 hit at Phase:88 Delta:216 Distance:0.16
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(X, 0.16, 25.00)
<<< do_homing_move  X0.00 Y0.00 Z5.00
>>> set_axis_is_at_home(X)
Axis 88 home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z5.00 : 
<<< set_axis_is_at_home(88)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(88)
>>> homeaxis(89)
Home Fast: -348.00mm
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(Y, -348.00, [50.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z5.00
Endstop 89 hit at Phase:664 Delta:232 Distance:0.17
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(Y, 0.17, 25.00)
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z5.00
>>> set_axis_is_at_home(Y)
Axis 89 home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X0.00 Y0.00 Z5.00 : 
<<< set_axis_is_at_home(89)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(89)
>>> home_z_safely  X0.00 Y0.00 Z5.00
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  destination= X160.00 Y114.00 Z5.00 : home_z_safely
>>> do_blocking_move_to  X0.00 Y0.00 Z5.00
>  X160.00 Y114.00 Z5.00
echo:busy: processing
<<< do_blocking_move_to  X160.00 Y114.00 Z5.00
>>> homeaxis(90)
  current_position= X160.00 Y114.00 Z5.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X160.00 Y114.00 Z5.00
>  X160.00 Y114.00 Z10.60
<<< do_blocking_move_to  X160.00 Y114.00 Z10.60
>>> do_blocking_move_to  X160.00 Y114.00 Z10.60
>  X160.00 Y114.00 Z10.60
<<< do_blocking_move_to  X160.00 Y114.00 Z10.60
Home Fast: -384.00mm
>>> do_homing_move  X160.00 Y114.00 Z10.60
...(Z, -384.00, [4.00])
[... 126 lines cut ...]
echo:busy: processing
<<< do_homing_move  X160.00 Y114.00 Z10.60
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = -0.60
Axis 90 home_offset = 0.25 position_shift = 0.00
> home_offset[Z] = 0.25
  current_position= X160.00 Y114.00 Z0.60 : 
<<< set_axis_is_at_home(90)
  current_position= X160.00 Y114.00 Z0.60 : sync_plan_position
  current_position= X160.00 Y114.00 Z0.60 : > AFTER set_axis_is_at_home
  current_position= X160.00 Y114.00 Z0.60 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X160.00 Y114.00 Z0.60
>  X160.00 Y114.00 Z0.60
<<< do_blocking_move_to  X160.00 Y114.00 Z0.60
<<< homeaxis(90)
<<< home_z_safely  X160.00 Y114.00 Z0.60
>>> do_blocking_move_to  X160.00 Y114.00 Z0.60
>  X160.00 Y114.00 Z5.00
<<< do_blocking_move_to  X160.00 Y114.00 Z5.00
  current_position= X160.00 Y114.00 Z5.00 : sync_plan_position
X:160.00 Y:114.00 Z:5.25 E:0.00 Count X:12800 Y:9120 Z:2000
<<< G28  X160.00 Y114.00 Z5.00
ok

thinkyhead, 2021-04-07:
Once X and Y are homed they will retain their positions and won't get reset to their home positions unless homing moves are being attempted. It's possible that the X and Y sensorless states is getting triggered early when starting from certain positions, or the triggered state is somehow "stuck." See if it helps to lower your max acceleration for X and Y — starting with something very low like `M201 X200 Y200` before attempting the second homing.

kpishere, 2021-04-08:
Have had issue of sensorless homing being too sensitive.  I have reduced it a while ago. Also, I'd turned on monitoring of end stops to see if they were triggering -- they weren't.  The associated PR (#21558) does fix the problem.  earlier on in the execution of G28, there appears to be checks that will cause homing to be skipped -- something in that logic isn't right for this case?  I think i did notice that Y axis hits twice now on first G28 call but I didn't notice that for X axis.

thinkyhead, 2021-04-13:
G28 will always re-home all the axes. There simply is no logic in Marlin that says "ignore G28" under any conditions.

kpishere, 2021-04-13:
Here is an image of the comparison, may be easier to spot.

home1 is output from first G28, home2 is output from second G28

[image]

thinkyhead, 2021-04-14:
I see nothing unusual going on there. After a homing move where the endstop gets triggered, the axis is set to the home position. From the log it seems that Marlin tries to home X, immediately sees the endstop triggered, stops the homing move, and sets the position to 0. There is no reason why Marlin would skip trying to move X and Y towards their endstops, unless those endstops are already triggered, for some reason.

Please do a test where you turn off `DISABLE_X`, `DISABLE_Y`, and `DISABLE_Z` to see whether this has any effect. These flags will affect the behavior of homing. However, according to my reading they should not cause the problem you are seeing. Still, we need to eliminate the possibility of a logical error in some other region that we've not yet looked at.

closed 2021-06-05 by thinkyhead