MarlinFirmware/Marlin #21524: [BUG] FLSUN HISPEEP Stuck in boot loop with SD inserted

Savage79, 2021-04-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Newest Marlin Bugfix downloaded April 2/2021.
Changed Env. to accommodate FLSUN_Hispeed motherboard.
Compiles no errors, copied .bin file to SD card, updates firmware, then gets stuck on Marlin boot screen, reboots and repeats. 

### Bug Timeline

Unknown

### Expected behavior

Expected to finish update cycle with SD card inserted.

### Actual behavior

Boot loop with SD card left in,  NO SD card will boot as required

### Steps to Reproduce

1. compiled firmware as I normally do.
2. copy Robin_mini.bin to SD card and proceeded to update motherboard
3. get stuck on Marlin Boot screen, last 10 seconds, reboots and repeats

### Version of Marlin Firmware

Marlin Bugfix 2.0.X

### Printer model

Flsun QQS Pro

### Electronics

Flsun HISPEED V1

### Add-ons

TMC2208 in UART

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

If SD card is left in the motherboard it will remain in a boot-loop, however 
when card is removed boots as normal.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6251594/Marlin.zip)

chrisjenda, 2021-04-03:
I have tracked commit 6e0b79a to be the root cause of this issue.
Pulling the latest commit of Bugfix Branch "Bump distribution date (2021-04-03)" (dd76a50) and reverting the commit fixes the issue.

rhapsodyv, 2021-04-03:
@chrisjenda thanks! I was suspecting of this too. But I didn’t had time to check.

If you don’t mind to test, I can send you a PR.

Probably it’s failing to use 4 bit.

chrisjenda, 2021-04-03:
I am 100% free to test anything to get this to work!  I give that PR a go!

Foxies-CSTL, 2021-04-05:
Hello,
I also had this problem that I solved by commenting in the line "//#define SDIO_READ_RETRIES 16" of the PINS file.
I have not revalidated it since.

petergebruers, 2021-06-25:
I can confirm the problem (branch bugfix-2.0.x Commit 0bd113b9) causes a boot loop on my QQ-S Pro if I leave the SD card in the printer. Tried 2 different cards 4GB white label vs Samsung 32 GB.

> I solved by commenting in the line "//#define SDIO_READ_RETRIES 16" of the PINS file.

I can confirm that this fixes the boot loop for me as well.

I only started using Marlin a week ago on my first 3d printer, I cannot add much weight to this discussion. I do toy with MCUs and I am wondering why "16" instead of "3" causes is issues, or why it was set to 16 in the first place (I ran git blame, does not give a hint). So consider this to be "a vote up" for that solution from "an informed noob"

dredivan, 2021-11-18:
I have same problem on BOARD_FLSUN_HISPEED FLSUN HiSpeedV1 STM32F103VET6 with latest bug-fix

`//#define SDIO_READ_RETRIES                   16`
don`t help me

tpruvot, 2021-11-18:
You can try to reduce the clock here : 
Marlin/src/pins/stm32f1/pins_FLSUN_HISPEED.h:258:  #define SDIO_CLOCK                     4500000  // 4.5 MHz

But your case may be due to the quality of your SD card or just to some errors on it, in general a fresh format fixes the problem too...

a common cheap SD card (not latest ones) can run between 18 and 20MHz if properly wired (not on a long flat cable without shielding)

closed 2023-10-19 by github-actions[bot]