MarlinFirmware/Marlin #21539: [BUG] LPC1768 Compile error with software end-stops enabled - PLATFORMIO/VSC/Win10

rrroonn, 2021-04-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Configuration.h:
// Max software endstops curtail movement above maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS 1
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X  305
  #define MAX_SOFTWARE_ENDSTOP_Y  205
  #define MAX_SOFTWARE_ENDSTOP_Z  310
#endif

Compile Result:

Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\tft\../../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
Marlin\src\HAL\LPC1768\tft\../../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
==================================================== [FAILED] Took 3.94 seconds ==================================================== 

Environment    Status    Duration
[... 85 lines cut ...]
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\tft\../../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
Marlin\src\HAL\LPC1768\tft\../../../inc/SanityCheck.h:648:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  648 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
==================================================== [FAILED] Took 3.94 seconds ==================================================== 

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:03.941
```
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/6255865/Marlin_config.zip)

I have attached my config files for reference.

thisiskeithb, 2021-04-05:
From the pasted log: `#error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."`.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2021-04-05 by thisiskeithb

rrroonn, 2021-04-05:
Thanks for looking at this.  Not sure why you closed it.  You can see that I DID ENABLE MAX_SOFTWATE_ENDSTOP_X, MAX_SOFTWATE_ENDSTOP_Y and MAX_SOFTWATE_ENDSTOP_Z, so there appears to be a logic error in the code which is what I am reporting.  Can you please look at that or let me know how I am wrong.

thank you

Ron

borland1, 2021-04-05:
You are not following the logic correctly.

You can only enable your endstops like this.  

```
// Max software endstops curtail movement above maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X 
  #define MAX_SOFTWARE_ENDSTOP_Y
  #define MAX_SOFTWARE_ENDSTOP_Z
#endif
```

You would set the build area limits in file `Configuration.h` under `@section machine` 

```
// @section machine

// The size of the print bed
#define X_BED_SIZE 305
#define Y_BED_SIZE 205

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 310
```

rrroonn, 2021-04-05:
Thank you very much for a helpful reply - that works.

thanks again,

Ron