MarlinFirmware/Marlin #21542: [BUG] Can't compile since 'Fix CardReader string args' commit

benlye, 2021-04-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Can't compile Marlin since https://github.com/MarlinFirmware/Marlin/commit/98d6c751d3484b1a3d30cefaa692b4f993d81abd

### Bug Timeline

New bug, started since 'Fix CardReader string args' commit six hours ago

### Expected behavior

Marlin should compile

### Actual behavior

Compile fails with this message:
```
Compiling .pio\build\LPC1769\src\src\sd\cardreader.cpp.o
Marlin\src\sd\cardreader.cpp: In static member function 'static void CardReader::write_command(const char*)':
Marlin\src\sd\cardreader.cpp:726:17: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
  726 |   char *begin = buf;
      |                 ^~~
      |                 |
      |                 const char*
Marlin\src\sd\cardreader.cpp:728:33: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
  728 |   char *end = buf + strlen(buf) - 1;
      |               ~~~~~~~~~~~~~~~~~~^~~
      |                                 |
      |                                 const char*
*** [.pio\build\LPC1769\src\src\sd\cardreader.cpp.o] Error 1
```

Reverting the commit solves the problem and Marlin compiles successfully.

### Version of Marlin Firmware

Bugfix 2.0.x

Foxcor, 2021-04-05:
I confirm.
If you uncomment #define SDSUPPORT, you get a compilation error.

valemaio2, 2021-04-05:
I can confirm too. Reverting the changes from 98d6c75 lets me compile again.

alexonpeace, 2021-04-05:
same here, reverting the changes did let me compile too

sidneychurch, 2021-04-05:
I was able to fix this issue by opening Marlin-bugfix-2.0.x\Marlin\src\sd\cardreader.cpp and changing:
 void CardReader::write_command(char * const buf)  to
void CardReader::write_command(char * const buf)

This should be on line 725 or 726. This also means that we need to change the function in the cardreader header, so now open Marlin-bugfix-2.0.x\Marlin\src\sd\cardreader.h and change:

static void write_command(const char * const buf); to 
static void write_command(char * const buf);

This should be around line 95 in the header. After changing these, it should compile.

ellensp, 2021-04-06:
this commit https://github.com/MarlinFirmware/Marlin/commit/e3116eed2e488543766cd61e4658363a0d2bd34f should fix it.

thisiskeithb, 2021-04-06:
> this commit e3116ee should fix it.

Fix it on my builds 🙂

closed 2021-04-06 by thisiskeithb