MarlinFirmware/Marlin #21543: [BUG] Marlin Freezes

hpmax, 2021-04-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Unfortunately, I don't know for sure this is Marlin, since I don't really know what's happening.

I have an FLSUN QQ printer which has an SKR 1.3 (TMC2209 drivers) running Marlin 2.0.x (nightly build, from, I think March 27th).  The printer appears to work fine.  I was able to PID Autotune the hot end and the bed, although (the numbers seem to bounce around considerably from one run to the next).  I was able to auto-cal and then auto-level with UBL.  I can manually move the hot end, turn the extruder in both directions, etc.

When I attempt to print via the included SD interface (which is coming from a separate board in the QQ) which I believe issues GCode over a serial port to the SKR.  The printer starts printing, but invariably stops after some time before the print completes.  When it stops, the hot-end, bed, and fan all stay on and set at the desired temperature.  I believe (but am not positive) that I am seeing the temperature displayed -- which indicates the SKR is responding to requests for temperature).

I have two options that I can do: "pause" which doesn't seem to do anything, or "stop" which causes the print head to home at which point I'm done.

My first thought was perhaps the G-Code is bad, or that my SD card is faulty and reading from it sometimes gets an error.  I printed the same object on another printer (FLSun Q5) from the same SD card (minus the uSD->SD sleeve) and it printed fine though, so I tend to think it's not that.

The history of this is complicated.  I had barely used the printer after switching to the SKR, and had a lot of trouble configuring Marlin (although I think it's now properly configured).  I just changed the auto-level sensor (used to be based on nozzle contact/conduction, now based on a switch).  I don't think I had seen the problem prior to switching to the switch, but at the same time, I can't imagine how that could be responsible for the problem.

I tried to log with M111 S247 but nothing came out the serial port to my computer.
I also tried M928 log.txt but nothing appeared to be written to either SD card (the one in the SKR, or the one in the board that controls the display)

I was able to replicate while watching it.  Absolutely nothing seemed to happen, it just got stuck.  Nothing came out of the log (when connected via Repetier).  However, I was able to issue it commands via Repetier -- it's conceivable to me that Marlin is only issuing debug information to the interface where it's receiving instructions (which is not Repetier).  Makes me suspect the problem may be the interface board.  I also noticed that the USB connector on the board seemed loose (I could make/break contact by jiggling the connector), and it happened on two different cables.

As a side note, using Repetier I was able to issue a command to go beyond the geometry of printer (Z+420mm, where actual limit is about 378mm), and it not only was willing to do that, but it appeared to ignore the endstop.  Is that a configuration issue?

### Bug Timeline

New issue (see description)

### Expected behavior

I expect prints to complete

### Actual behavior

Print did not complete

### Version of Marlin Firmware

bugfix 2.0.x, Nightly build March 27th, 2021

### Printer model

FL Sun QQ

### Electronics

SKR 1.3 w/ TMC2209, BMG-style extruder 

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

[Lid_cura.zip](https://github.com/MarlinFirmware/Marlin/files/6259129/Lid_cura.zip)
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6259134/Config.zip)

thisiskeithb, 2021-04-06:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

hpmax, 2021-04-06:
Thanks for the response.   I am using bugfix-2.0.x nightly build from March 27th.  Do you have reason to believe a newer release would fix this?  

I'm suspecting this may be an issue with my MKS TFT32.  After it locked up, I shut it off, and turned it back on and it allowed me to restart.   The question is why is the TFT32 freezing up is it because of a problem with the TFT32 or because Marlin isn't responding the way it expects (which might still be an issue with the TFT32 or might be an issue with Marlin).

thisiskeithb, 2021-04-06:
> Thanks for the response. I am using bugfix-2.0.x nightly build from March 27th. Do you have reason to believe a newer release would fix this?

The `bugfix-2.0.x` branch is updated daily, so that's why we always ask users to try the latest `bugfix-2.0.x`.

I haven't seen an MKS TFT lock up due to Marlin. I assume you're running MKS's firmware? Are you running the latest version? Do you have another TFT or LCD controller you can swap in to help with debugging?

hpmax, 2021-04-06:
So here's the thing.   I switched to Marlin after replacing the mainboard with the SKR 1.3.  I think I've done a couple small prints on it since replacing the mainboard. The MKS TFT32 was completely original, not updated. 

This morning I updated to the latest firmware (3.0.5, or is it 3.0.6) on the TFT32.  When I did, I noticed that the first setting in the config file is the firmware version (Repetier, Marlin, Smoothie).  I think the QQ originally ran Repetier, so it's possible that the TFT32 was configured to talk to Repetier instead of Marlin.  Either way,  I updated it and re-ran before leaving for work.   So when I get home I'll see if it fixes the problem.  It's always possible the SD card sleeve is faulty although I'm pretty confident the uSD is working. 

If this doesn't work,  I'll update to the latest build and I'll try to push the print through USB on my computer.

thisiskeithb, 2021-04-06:
> So here's the thing. I switched to Marlin after replacing the mainboard with the SKR 1.3. I think I've done a couple small prints on it since replacing the mainboard. The MKS TFT32 was completely original, not updated.

You likely have the incorrect firmware type in the TFT firmware. For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-06 by thisiskeithb

hpmax, 2021-04-07:
After reflashing the TFT32 firmware I made a dry print and it completed, which seems to be a "success."  Will try two or three more before declaring this resolved.