MarlinFirmware/Marlin #21547: [BUG] TEENSY4.1 and Laser control problem

msnasel, 2021-04-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I uncomment #define LASER_FEATURE in I can't compile Marlin

### Bug Timeline

new

### Expected behavior

-

### Actual behavior

-

### Version of Marlin Firmware

bugfix-2.0.x

> Executing task: C:\Users\Martin\.platformio\penv\Scripts\platformio.exe run --environment teensy41 <

Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.12.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.153.0 (1.53)
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 87 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <PWMServo> 2.1
|-- <SD> 1.2.2
|   |-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <Time> 1.6
|-- <Teensy_ADC> 8.0.38
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\teensy41\src\src\MarlinCore.cpp.o
Compiling .pio\build\teensy41\src\src\feature\spindle_laser.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pio\build\teensy41\src\src\gcode\control\M211.cpp.o
In file included from C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/WProgram.h:45:0,
                 from C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/Arduino.h:6,
                 from Marlin\src\inc/../HAL/./TEENSY40_41/../shared/Marduino.h:36,
                 from Marlin\src\inc/../HAL/./TEENSY40_41/HAL.h:30,
                 from Marlin\src\inc/../HAL/HAL.h:26,
                 from Marlin\src\inc/MarlinConfig.h:31,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/wiring.h:131:18: error: statement-expressions are not allowed outside functions nor in template-argument lists
 #define round(x) ({ \
                  ^
Marlin\src\feature/spindle_laser.h:49:15: note: in expansion of macro 'round'
     min_pct = round(TERN(CUTTER_POWER_RELATIVE, 0, (100 * float(SPEED_POWER_MIN) / TERN(SPINDLE_FEATURE, float(SPEED_POWER_MAX), 100)))),
               ^
C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/wiring.h:131:18: error: statement-expressions are not allowed outside functions nor in template-argument lists
 #define round(x) ({ \
                  ^
Marlin\src\feature/spindle_laser.h:50:15: note: in expansion of macro 'round'
     max_pct = round(TERN(SPINDLE_FEATURE, 100, float(SPEED_POWER_MAX)));
               ^
In file included from C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/WProgram.h:45:0,
                 from C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/Arduino.h:6,
                 from Marlin\src\feature\../inc/../HAL/./TEENSY40_41/../shared/Marduino.h:36,
                 from Marlin\src\feature\../inc/../HAL/./TEENSY40_41/HAL.h:30,
                 from Marlin\src\feature\../inc/../HAL/HAL.h:26,
                 from Marlin\src\feature\../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\spindle_laser.cpp:27:
C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/wiring.h:131:18: error: statement-expressions are not allowed outside functions nor in template-argument lists
 #define round(x) ({ \
                  ^
Marlin\src\feature\spindle_laser.h:49:15: note: in expansion of macro 'round'
     min_pct = round(TERN(CUTTER_POWER_RELATIVE, 0, (100 * float(SPEED_POWER_MIN) / TERN(SPINDLE_FEATURE, float(SPEED_POWER_MAX), 100)))),
               ^
C:\Users\Martin\.platformio\packages\framework-arduinoteensy\cores\teensy4/wiring.h:131:18: error: statement-expressions are not allowed outside functions nor in template-argument lists
 #define round(x) ({ \
                  ^
Marlin\src\feature\spindle_laser.h:50:15: note: in expansion of macro 'round'
     max_pct = round(TERN(SPINDLE_FEATURE, 100, float(SPEED_POWER_MAX)));
               ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\teensy41\src\src\feature\spindle_laser.cpp.o] Error 1
*** [.pio\build\teensy41\src\src\MarlinCore.cpp.o] Error 1

ellensp, 2021-04-06:
Cannot reproduce.
Please attach your configuration files and any other edited files.

msnasel, 2021-04-06:
I mistaken the build. It does not work on version 2.0.7.2.
bugfix works.

closed 2021-04-06 by msnasel