MarlinFirmware/Marlin #21549: creality upgrade touchscreen support

jmartens2, 2021-04-05:
### Is your feature request related to a problem? Please describe.

creality v4.2.7 board wont run abl with their ender3 pro upgrade touchscreen. ive gone thru marlin v2.0.7 back and forth i cannot find a way to enable the touchscreen (i can only get a bin file from seller) to start with. i have to use their files. well that leaves other features disabled. im totally a noob and im trying hard but after 2 days of google searches and posting at a couple of creality sites with no answers i thought i would try here. forgive me if im in wrong place.

### Are you looking for hardware support?

creality silent board v4.2.7
creality upgraded touchscreen for ender 3 pro

### Describe the feature you want

i would like to activate the touchscreen and have any features from ABL, Z-offset.
i would like to just be able to use the stock screen and run gcode in slicer before prints.
i cant figure out how to do this.

thisiskeithb, 2021-04-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2021-04-06 by thisiskeithb

jmartens2, 2021-04-06:
Thanks for reply but I thought that since the choices weren’t always available for feature them maybe they could be? Just thought I’d ask. Have a nice day.