MarlinFirmware/Marlin #21554: Marlin 2.0.7.2 locks Ender 3 MAX with Bltouch motors on startup.

oliverhbailey, 2021-04-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Updated Creality firmware 1.0.16-bltouch with Marlin 2.0.7.2 bugfix full version. On first boot, had to initialize EEPROM, which I did and receive message file had been written to EEPROM. Turned off printer for about a minute, then turned printer back on. Bltouch deployed and stowed twice, and I went to test manual motor movements. Chose axis-Z-1.00 MM and dialed down 0.100 and nothing happened. Eventually a stopped message appeared and requested a reset. Turn off and turned on again, Bltouch went through startup test and I chose a auto home, same end result, stopped message appeared and requested a reset.

A couple of notes:
This Ender 3 MAX was received in late February and has the 4.2.2 controller. The BLtouch was purchased from Amazon and is the official Creality Bltouch for the Ender series.  Having said that however, the latest firmware from Creality would not store or allow the Z axis to move to the build plate. I would always generate an "Err Too Far" and fail to locate the Z axis below the 10mm space between the bed and the Bltouch. 

When compiling the Marlin 2.0.7.2, I used the modification instructions from 2.0.3 for setting the Bltouch defines. There is no documentation on the Ender 3 MAX specifically but Marlin appears to be configured for the 4.2.2/4.2.7 controller. I am a 50 year hardware and embedded software developer, so I can delve deeper into this if you need more answers.

The Stepper Motors were locked from power on.

### Bug Timeline

This is a new build for the Ender 3 MAX for me.

### Expected behavior

I dialed the motion selector down (Counter Clock Wise) and expected immediate motor response. I brief hum followed by "STOPPED" message followed by "RESET CONTROLLER" message. The stepper motors  were locked at startup.

### Actual behavior

Nothing moved, Bltouch initialized at powerup. Motors were locked at powerup and remained so until power off.

### Steps to Reproduce

1. Flash Marlin 2.0.7.2 with Bltouch configured
2. Initialize EEPROM after Startu[
3. Turn unit off, wait 30 seconds, turn unit on.
4) Select prepare motion-x axis-1.0mm and turn knob coutnter-slock-wise to lower z axis.

### Version of Marlin Firmware

2.0.7.2 bugfix

### Printer model

Creality Ender 3 MAX

### Electronics

Stock Electronics, Creality Bltouch upgrade kit for Ender Series

### Add-ons

No other addons are installed.

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

[Marlin-2.0.7.2.zip](https://github.com/MarlinFirmware/Marlin/files/6266050/Marlin-2.0.7.2.zip)

Actually this is a 7z file, I had to name it because a 7z file cannot be dropped in this window.

thisiskeithb, 2021-04-06:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

oliverhbailey, 2021-04-06:
Hi Keith,
The Ender 3 MAX is not well described technically by Creality.
Documentation from Creality states it is a 4.2.7 board but their firmware
states it is a 4.2.2. I did not realize (until after I wrote the bug
report) that Marlin uses Ender 3 Pro 1.5 for the 4.2.2 Ender 3 MAX. I
downloaded, configured, compiled, and flashed the unit, and all is working
perfectly now. You may want to add to your Ender 3 Pro 1.5, "Ender 3 Pro /
Ender 3 MAX 1.5" to let anyone who is using the Ender 3  MAX with a 4.4.2
board they can use this. What kept it from working was the inverted control
pin on the Z axis driver.

Thank for for getting back so quickly Keith. Creality is really doing
themselves a big disservice by the issues they have and create with these
boards.

Again, thank you again.

Sincerely,
Oliver Bailey

PS - You and your associates have done a very good job with Marlin, I use
it customized, in all my machines.

closed 2021-04-07 by ellensp