MarlinFirmware/Marlin #21555: TMC5160 Uart Support

Maazshakir, 2021-04-06:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

I made a custom electronics with stm32 f4 chip , so its compatible with marlin

### Describe the feature you want

I would love to see a uart support for tmc5160 i have a board which doesnt support software spi  but is capable of uart mode

thisiskeithb, 2021-04-06:
@teemuatlut Does your library support UART for the 5160? Should this FR go there instead?

kpishere, 2021-04-09:
I don't get it. Why doesn't it support it? Lots of people use tmc2209's in uart mode. I use Serial2 w. arduino in hardware uart on a due and mega2560.  you may have to change more than config files to switch to that.

thisiskeithb, 2021-04-09:
The 5160 can communicate through SPI or UART, but defaults to SPI within Marlin (and that's why I tagged teemuatlut to see if this is a library or Marlin request).

Maazshakir, 2021-04-10:
In my opinion in marlin uart mode for tmc5160 is not supported yet but it
would be great if you can include it in nest update

Keith Bennett ***@***.***> ezt írta (időpont: 2021. ápr. 9.,
P, 4:06):