MarlinFirmware/Marlin #21557: MKS Robin Nano v2.0 and MKS TS35 V2.0 Icon Layout Issue

descipher, 2021-04-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When 2 extruders are defined the 2nd extruder icon overlaps with the tool icon

### Bug Timeline

Last 15 Days prior to 06/04/2021 

### Expected behavior

Icons spaced without overlap.

### Actual behavior

Photo included.

### Steps to Reproduce

Compile and boot with the new firmware and assets folder from the pio build dir. 

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

N/A

[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6268777/Configuration.zip)

Malderin, 2021-04-11:
@descipher , 
try this: https://github.com/MarlinFirmware/Marlin/pull/21595

thisiskeithb, 2021-04-12:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/21595. Closing.

closed 2021-04-12 by thisiskeithb