MarlinFirmware/Marlin #21559: [BUG] stepper not stopping after hitting endstop while homing

MelanieKahl, 2021-04-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I home an axis (happens on all three axes X, Y and Z) the motor moves until the endstop (mechanical endstop) is triggered, bumps back, but then the motor doesn't stop moving, but instead it moves again in the direction of the endstop, but slower this time a bit slower. After some seconds the printer halted ([ERROR] Error:Printer halted. kill() called!).

I checked the endstops with M119, all are working fine and are triggered, when they are pressed.

### Expected behavior

I expect the motor to stop, when the endstop is triggered while homing.

### Actual behavior

The motor does not stop moving, when endstop is triggered.

### Steps to Reproduce

1. Home any axis  (X, Y or Z)

### Version of Marlin Firmware

Marlin 2.0.7.2

### Electronics

RAMPS 1.6+, TMC2130 drivers

### Add-ons

Mechanical Endstop

### Host Software

Pronterface

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6272700/Configuration.zip)

borland1, 2021-04-09:
Sounds like a hardware problem to me.

You have configured for normally open (NO) endstops with inverting to TRUE (typically FALSE) and also configured for pullup resistors with `#define ENDSTOPPULLUPS`.  For RAMPS 1.4, endstops are usually configured for normally closed (NC), which is so that if any endstop's wire harness becomes disconnected, it triggers a stop (pulled high by pullup resistors).  Check individually with M119, how your endstop switches trigger with the associated harness disconnected.

Also, there could be a electrical noise problem when the board's microcontroller debounces the endstop mechanical switches.  There is an optional ENDSTOP_NOISE_THRESHOLD setting `Configuration.h` for adjusting that debounce period.  RAMPS 1.4 mechanical endstops typically include a circuit board with an added capacitor to help debounce the switch internal contacts.

fedetony, 2021-04-18:
Exact same issue here. I set the homing cycle in one axis (happens in all axis) I trigger the endstop manually before it bumps and the stepper keeps moving some additional 10-15mm after triggering, bumps back 5 mm and triggers Kill(). Also shows.
```
g28 X

S_XYZ:9
echo:busy: processing
X:-37.9950 Y:0.0000 Z:75.0000 E:0.0000 Count X:-7614 Y:0 Z:60000
S_XYZ:9
[MarlinConfig2.0.7.2.zip](https://github.com/MarlinFirmware/Marlin/files/6332218/MarlinConfig2.0.7.2.zip)

echo:busy: processing
X:-77.9550 Y:0.0000 Z:75.0000 E:0.0000 Count X:-15605 Y:0 Z:60000
S_XYZ:9
echo:Bad X Endstop?
Error:Printer halted. kill() called!
//action:poweroff
```

I am running the SKR1.4 with bugfix2.0.x 20008 here, but was happening as well in Marlin 2.0.7.2, version 200007.

fedetony, 2021-04-20:
@thisiskeithb is there any advancement on this issue? I have not been able to home my machine at all, I need to set the homing position manually with G92. :)

thisiskeithb, 2021-04-20:
> @thisiskeithb is there any advancement on this issue?

No. There’s quite the backlog: https://github.com/MarlinFirmware/Marlin/issues

MelanieKahl, 2021-04-23:
I somehow figured out that I there was problem with the line

#define HOMING_BUMP_MM      { 10, 10, 10 }  

in the configuration_adv.h file.

Inititally I wanted the stage move away from the endstop after homing, so that the endstop switch is not triggered anymore. Hence, I set the values quite high (around 50 - 100). But somehow with these values the motor doesn't stop moving, but with values around 10 the motor stops at the position, where it should stop. So now, the homing procedure works as it should.

thisiskeithb, 2021-04-23:
> So now, the homing procedure works as it should.

Great news!

closed 2021-04-23 by thisiskeithb

fedetony, 2021-04-23:
Mnn, if you check my config files, you'll notice my values are:
```
#define HOMING_BUMP_MM      { 10, 10, 5 }       // (mm) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 8, 8, 4 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)
#define HOMING_BACKOFF_POST_MM { 5, 5, 5 }  // (mm) Backoff from endstops after homing
``` 
what is the catch?