MarlinFirmware/Marlin #21561: [BUG] the wire pusher is bugged

GHOSTS15, 2021-04-07:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

in the example configuration for the geeetech A20 printer, the wire pusher works in reverse (instead of pushing it into the extruder it pulls it out)

### Bug Timeline

in marlin 2.0.7.2

### Expected behavior

that the wire pusher (I don't know the technical name) worked well 

### Actual behavior

that the wire pusher (I don't know the technical name) doesn't work well

### Version of Marlin Firmware

marlin 2.0.7.2

### Printer model

geeetech a20

### Your Slicer

Cura

### Host Software

Cura

ETE-Design, 2021-04-08:
Have you tryed to invert E0 in Configuration.h?

Vertabreak, 2021-04-08:
PR submitted to correct this.

GHOSTS15, 2021-04-08:
I have reported it so that you can correct it in the next stable versions

thisiskeithb, 2021-04-08:
> I have reported it so that you can correct it in the next stable versions

Thanks for the report!

@Vertabreak put in a pull request to have it fixed: https://github.com/MarlinFirmware/Configurations/pull/463

ellensp, 2021-04-08:
Closing issue, as this is for Configurations and PR has been created.

closed 2021-04-08 by ellensp