MarlinFirmware/Marlin #21563: [BUG] Issue compiling for TronXY X5SA-Pro

orangeworx, 2021-04-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to compile with #define TFT_LVGL_UI (using VS Code), the terminal output is attached as a text file (error log).

### Bug Timeline

Uncertain as it's my first build

### Expected behavior

Successful compilation

### Actual behavior

Failure to compile - #error xxxx requires a programmable LCD controler.

### Steps to Reproduce

1- Test build with Marlin 2.0 bug fix (latest) and standard config files
2- Swapping configuration.h and configuations_adv.h with the examples>TronXY>X5SA
3- Modifications following the README.MD file from X5SA examples folder.
4- Enablin TFT_LVGL_UI
5- Compiling

### Version of Marlin Firmware

bugfix-2.0.x (Mar 21 2021 19:56:35)

### Printer model

TronXY X5SA

### Electronics

Stock v6 board, TMC2208

### Add-ons

N/A

### Your Slicer

Cura

### Host Software

OctoPrint

Hello everyone!
I've followed the procedure in the README.MD/examples for the X5SA but for the X5SA-Pro. I've managed to go through all the issues that presented as I was moving forward.
But there's one that's got me scratching my head. The #define TFT_LVGL_UI option seems to fail at compiling. I've "seen" a couple of working setups:
- a couple of posts talking about issues with the LVGL but after successful compilation for a v6 board
- a youtube video where the OP is showing the functioning screen/GUI
so I must be missing something.

[error.log](https://github.com/MarlinFirmware/Marlin/files/6276080/error.log.txt)

Included config files (with #define TFT_LVGL_UI enabled)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6276084/config.zip)

But I've had a chance to test a couple of things out and managed to get a build to succeed albeit minus a couple of features...
Bizarrely, every time I ran the build with #define FILAMENT_RUNOUT_SENSOR or #define ADVANCED_PAUSE_FEATURE uncommented, it failed.

So with #define TFT_LVGL_UI enabled (uncommented), here are my results;
(for easier reading, I'm using enabled for uncommented, disabled for commented.)
1- Compiles only if both are enabled.
2- Fails when ADVANCED_PAUSE_FEATURE is enabled, FILAMENT_RUNOUT_SENSOR is disabled
VSCode Terminal shows a repeat of the following error
SanityCheck.h:872:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)." (in red text)
#error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."(in yellow text)
Enabling EMERGENCY_PARSER, also fails (seems to be another issue altogether, something about pause_menu_response). I've attached the logs for that separately.
[EP1APF1FRS1.log](https://github.com/MarlinFirmware/Marlin/files/6276098/EP1APF1FRS1.log)
3- Fails if FILAMENT_RUNOUT_SENSOR is enabled and ADVANCED_PAUSE_FEATURE disabled
VSCode Terminal shows a repeat of the following error
SanityCheck.h:863:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR. (in red text)
static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR."); (in yellow text)

Hope this sheds some light. If anyone has any ideas, I don't mind testing.

Vertabreak, 2021-04-08:
sanity checks are telling you ADVANCED_PAUSE_FEATURE & LCD_BED_LEVELING are not compatible with your screen.

orangeworx, 2021-04-08:
I believe I understand what the sanity checks do though, in this instance, they're triggered when TFT_LVGL_UI is enabled... these errors don't appear when I have #define TFT_COLOR_UI enabled...
I forgot to mention that I had compiled the firmware with same settings, only UI difference. no issues with screen. I wanted to test out the other UI out of curiosity and seeing how some folks had successfully built it.

Vertabreak, 2021-04-08:
it is possible they disabled the sanity checks to bypass them but sanity checks are put in place for good reasons,

closed 2021-04-08 by thisiskeithb