MarlinFirmware/Marlin #21565: Marlin config file for Geetech A10T is not correct.  Please check

ace1966, 2021-04-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin does not display on screen.   Printing form USB cable only.

### Bug Timeline

No

### Expected behavior

None

### Actual behavior

Prints only with USB cable.  No wording on LCD, but it does lite up.

### Steps to Reproduce

Tuen the machine,  
No wording comes on but screen lites up.

### Version of Marlin Firmware

Bugfix 2.0x

### Printer model

Geetech A10T

### Electronics

Stock

### Add-ons

None

### Your Slicer

Cura

### Host Software

Cura

I have tried installing the firmware upgrade for the A10T  with no luck.  The config file I don't believe contains the correct info.
After installing the firmware the LCD screen lights up but contains no wording.   The printer will print from USB cable only.

I have not made any modifications to this printer except for trying to upgrade firmware.

Vertabreak, 2021-04-08:
Did you use the base version config or the yhcb version if you have a v4.1 board you need yhcb.

ace1966, 2021-04-08:
Can I reinstall it now with the correct version?

Vertabreak, 2021-04-08:
Switch from discount reprap lcd to yhcb disable one enable other reflash. Both examples are provided in examples.

ace1966, 2021-04-08:
Sorry I am new to 3d printing,  could you tell me where I can find the yhcb versionthanks for your fast response.

Vertabreak, 2021-04-08:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Geeetech/A10T

ace1966, 2021-04-08:
I can't get any of the A10T config files to work, they all come up with error message.I guess I'm too old for this stuff (73)Thanks for Your help.

 
 

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Geeetech/A10T

Vertabreak, 2021-04-08:
what error?

ace1966, 2021-04-08:
Thanks for all your help, I think I am getting it now.

Vertabreak, 2021-04-08:
if the issue is solved you can close the issue.

thisiskeithb, 2021-04-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2021-04-08 by thisiskeithb

ace1966, 2021-04-08:
Sorry new to the site.

ace1966, 2021-04-09:
I really appreciated your help on getting my 3d printer back to working.   I would of never gotten it working again without your help.Thanks