MarlinFirmware/Marlin #21570: [BUG] Failed compilation

orangeworx, 2021-04-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Both default and X5SA (modded for pro) configurations fail compilation
bugfix 2.0.x downloaded yesterday, examples for 2.0.x downloaded yesterday.

### Bug Timeline

yesterday

### Expected behavior

successful compilation

### Actual behavior

failure

### Steps to Reproduce

download marlin 2.0.x bug fix
load folder in VSCode, run default build > failure, clean and reattempt > failure
download examples and mod the config for X5SA-Pro by following directions
load folder in VSCode, run build for Chitu_v6 > failure, clean and reattempt > failure

### Version of Marlin Firmware

bugfix-2.0.x (Mar 21 2021 19:56:35)

### Printer model

TronXY X5SA-Pro

### Electronics

Stock v6 board

### Add-ons

N/A

### Your Slicer

Cura

### Host Software

OctoPrint

Here are the files.
[Config](https://github.com/MarlinFirmware/Marlin/files/6280575/Marlin2.0.xBF-X5SA-Pro.zip)

[Error Log](https://github.com/MarlinFirmware/Marlin/files/6280578/failedCompilation.txt)

These errors only started when the marlin files were updated and are pretty much out of the box considering I haven't done anything other than following the guide (which worked previously, previous build is running ok and printing atm)

thanks!

Vertabreak, 2021-04-08:
i successfully compiled the example provided on the config repo.
after that i used your configs posted and also successfully compiled.
default_envs = chitu_f103

orangeworx, 2021-04-08:
then I'm at a loss... I've not changed anything on my system aside from flushing the BF files with the ones from the archive I downloaded yesterday... what else could it be? and what about the errors?

Vertabreak, 2021-04-08:
i had no errors, a few warnings but nothing that prevents compile.

orangeworx, 2021-04-08:
does my error log indicate anything specific?
I still can't compile... created a new folder with fresh everything and still.... same errors

Vertabreak, 2021-04-08:
> Executing task: platformio run <

Processing chitu_f103 (platform: ststm32@~12.0; board: CHITU_F103; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/CHITU_F103.html
PLATFORM: ST STM32 (12.0.0) > CHITU STM32F103Z (64k RAM. 512k Flash)
HARDWARE: STM32F103ZET6 72MHz, 59.12KB RAM, 469.03KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/board.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/wirish/boards.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/wirish/start.S.o
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/wirish/start_c.c.o
Compiling .pio/build/chitu_f103/FrameworkArduinoVariant/wirish/syscalls.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/Print.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/WString.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/cxxabi-compat.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/ext_interrupts.cpp.o
Compiling .pio/build/chitu_f103/FrameworkArduino/hooks.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/itoa.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/adc.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/adc_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/bkp_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/dac.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/dma.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/dma_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/exc.S.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/exti.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/exti_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/flash.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/fsmc_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/gpio.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/gpio_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/i2c.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/i2c_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/iwdg.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/nvic.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/pwr.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/rcc.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/rcc_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/spi.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/spi_f1.c.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/stm32f1/performance/isrs.S.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/stm32f1/performance/vector_table.S.o
Compiling .pio/build/chitu_f103/FrameworkArduino/libmaple/systick.c.o
[... 389 lines cut ...]
Compiling .pio/build/chitu_f103/lib87b/USBComposite/usb_setup.cpp.o
Compiling .pio/build/chitu_f103/lib87b/USBComposite/usb_x360_generic.c.o
Compiling .pio/build/chitu_f103/lib87b/USBComposite/usb_x360w.c.o
Compiling .pio/build/chitu_f103/libcda/Servo/Servo.cpp.o
Archiving .pio/build/chitu_f103/lib87b/libUSBComposite.a
Indexing .pio/build/chitu_f103/lib87b/libUSBComposite.a
Archiving .pio/build/chitu_f103/libcda/libServo.a
Indexing .pio/build/chitu_f103/libcda/libServo.a
Linking .pio/build/chitu_f103/firmware.elf
Building .pio/build/chitu_f103/firmware.bin
Checking size .pio/build/chitu_f103/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.1% (used 15184 bytes from 60536 bytes)
Flash: [====      ]  39.5% (used 189500 bytes from 480288 bytes)
encrypt([".pio/build/chitu_f103/firmware.bin"], [".pio/build/chitu_f103/firmware.elf"])
Block Count is  94
================================================================================== [SUCCESS] Took 38.45 seconds ==================================================================================

Environment    Status    Duration
-------------  --------  ------------
chitu_f103     SUCCESS   00:00:38.452
=================================================================================== 1 succeeded in 00:00:38.452 ===================================================================================

Terminal will be reused by tasks, press any key to close it.
[Marlin-bugfix-2.0.x-X5SA.zip](https://github.com/MarlinFirmware/Marlin/files/6280852/Marlin-bugfix-2.0.x-X5SA.zip)

orangeworx, 2021-04-08:
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS C:\Marlin-bf2> platformio run -e chitu_f103
Processing chitu_f103 (platform: ststm32@~12.0; board: CHITU_F103; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/CHITU_F103.html
PLATFORM: ST STM32 (12.0.0) > CHITU STM32F103Z (64k RAM. 512k Flash)
HARDWARE: STM32F103ZET6 72MHz, 59.12KB RAM, 469.03KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\chitu_f103\src\src\lcd\dogm\ultralcd_DOGM.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_bed_corners.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_bed_leveling.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_custom.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_filament.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_game.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_main.cpp.o
In file included from c:\marlin-bf2\marlin\src\lcd\marlinui.h:28:0,
                 from Marlin\src\lcd\dogm\../menu/menu.h:24,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
c:\marlin-bf2\marlin\src\lcd\buttons.h:221:0: warning: "LCD_BACK_CLICKED" redefined
   #define LCD_BACK_CLICKED() (buttons & EN_D)
 
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:223:0: note: this is the location of the previous definition
   #define LCD_BACK_CLICKED() false
 
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
c:\marlin-bf2\marlin\src\lcd\marlinui.h:79:0: warning: "LCD_UPDATE_INTERVAL" redefined
   #define LCD_UPDATE_INTERVAL TERN(HAS_TOUCH_BUTTONS, 50, 100)
 
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:77:0: note: this is the location of the previous definition
   #define LCD_UPDATE_INTERVAL TERN(HAS_TOUCH_XPT2046, 50, 100)
 
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_media.cpp.o
Compiling .pio\build\chitu_f103\src\src\lcd\menu\menu_mixer.cpp.o
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
c:\marlin-bf2\marlin\src\lcd\marlinui.h:67:8: error: multiple definition of 'enum LCDViewAction'
   enum LCDViewAction : uint8_t {
        ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:65:8: note: previous definition here
   enum LCDViewAction : uint8_t {
        ^~~~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
[... 20 lines cut ...]
c:\marlin-bf2\marlin\src\lcd\marlinui.h:88:19: error: redefinition of 'void wrap_string(uint8_t&, uint8_t&, const char*, bool)'
       inline void wrap_string(uint8_t &col, uint8_t &row, const char * const string, const bool wordwrap=false) { _wrap_string(col, row, string, read_byte_ram, wordwrap); }
                   ^~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:85:17: note: 'void wrap_string(uint8_t&, uint8_t&, const char*, bool)' previously defined here
     inline void wrap_string(uint8_t &col, uint8_t &row, const char * const string, const bool wordwrap=false) { _wrap_string(col, row, string, read_byte_ram, wordwrap); }
                 ^~~~~~~~~~~
In file included from Marlin\src\lcd\dogm\../menu/menu.h:24:0,
                 from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:293:
c:\marlin-bf2\marlin\src\lcd\marlinui.h: At global scope:
c:\marlin-bf2\marlin\src\lcd\marlinui.h:99:8: error: multiple definition of 'enum MarlinFont'
   enum MarlinFont : uint8_t {
        ^~~~~~~~~~
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:52:0:
Marlin\src\lcd\dogm\../ultralcd.h:235:8: note: previous definition here
   enum MarlinFont : uint8_t {
        ^~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\chitu_f103\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
====================================================================================================== [FAILED] Took 20.56 seconds ======================================================================================================

Environment    Status    Duration
-------------  --------  ------------
chitu_f103     FAILED    00:00:20.564
================================================================================================= 1 failed, 0 succeeded in 00:00:20.564 =================================================================================================