MarlinFirmware/Marlin #21572: [BUG] Latest 2.0 Bugfix Branch just compiled now BLTOUCH menu disappear on MKS UI version

BaltasarParreira, 2021-04-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

As the title says, the bltouch menu disappear, so now it's difficult to calibrate after the autolevel the z-offset.
This version I just compiled now fixes a lot of problems I have before, but this is really important to easy tune that parameter.
Using just by setup a value on the level settings menu it's a shot in the dark since I can't see if the value I'm putting is ok or going to have my nozzle hitting the bed making a dent !

### Bug Timeline

Just now after I compile and use the corrent 2.0 Bugfix Branch

### Expected behavior

Like before, after choosing auto level button, and after finishing the probe of all the bed points setup from the Configuration.h file the bltouch menu shoes up so we have the possibility to manual check and save the Z-offset correct amount.

### Actual behavior

The reported bltouch menu is not show anywhere.

### Steps to Reproduce

1- Choose Auto Level
2- After all the bed points are checked

### Version of Marlin Firmware

2.0 Bugfix Branch from today with MKS UI settings

### Printer model

Two Trees Bluer

### Electronics

MKS Robin Nano 1.2 with 3.5" touch TFT

Malderin, 2021-04-11:
I do not understand what kind of bltouch menu are you talking about?

BaltasarParreira, 2021-04-11:
You press on auto level button so the printer does the auto probing points (default 3 x 3 if we didn't change it at Configuration.h).
After finished that should show up another menu where you can them move de Z manually until the nozzle hits the piece o paper that we usually use to adjust the right Z offset value. Now there is no way of doing this seeing when the nozzle is at the right point. Just going to the settings menu a put a value on z-offset is a shot in the dark that afterword's when printing could make the nozzle hit the bed because was a guessed value.
Do you understand now?

This is the menu that I'm talking about and existed before:
https://user-images.githubusercontent.com/8551213/112407070-b676ca00-8d15-11eb-96b4-e8a6d2464553.png

Malderin, 2021-04-11:
@BaltasarParreira 
Of course, I could be wrong, but I have not seen such a menu in the official marlin. I saw such a menu in the MKS branch and not here.

BaltasarParreira, 2021-04-11:
So this branch even being more advanced and with some important bugs fixed is no good for using to compile a firmware using the MKS UI?
The real MKS you are saying has WIFI problems like for me just having the WIFI module connected makes the printer reboot once in a while or on uploading files to print because of some watchdog bugs. This one here the WIFI works perfect, but very difficult to calibrate the Z offset using Bltouch as the only way is get the value on trial and error.
You can't even go to the moving axis menu and lower the Z until you see the value you want, take a note a them go a put it in the settings because when reaches zero it stops and doesn't allow you to go lower.

Malderin, 2021-04-11:
@BaltasarParreira 
Hopefully @MKS-Sean  will someday transfer these additions to the official marlin. For now, you will have to use other less convenient methods to configure the bltouch.

thisiskeithb, 2021-04-12:
Closing since this is about MKS’ fork which includes features not implemented here.

closed 2021-04-12 by thisiskeithb