MarlinFirmware/Marlin #21576: [BUG] Ender 5 Plus `bugfix-2.0.x` code compiles with errors in VS Code

TheAZtech, 2021-04-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am attempting to compile the `bugfix-2.0.x` code using the Ender 5 Plus configuration files and it dumps several errors in the process in VS Code. I was able to successfully compile the `bugfix-2.0.x` code with the stock configuration files, but when adding the Ender 5 Plus files, it fails. The following screen shots are the locations of the code that is throwing the errors. 

[image]

[image]

### Bug Timeline

Unknown. 

### Expected behavior

I expected a successful compilation. :-)

### Actual behavior

The compilation failed. 

### Steps to Reproduce

Open the folder where `bugfix-2.0.x` code is stored; replace the configuration files from the 'Configurations-bugfix-2.0.x' file for the 'Ender-5 Plus'. Open folder in VS Code and click the hammer... Failed.

### Version of Marlin Firmware

Marlin `bugfix-2.0.x` code

### Printer model

Ender 5 Plus

### Electronics

Creality Silent Board V2.2

### Add-ons

No

### Your Slicer

Cura

### Host Software

OctoPrint

Vertabreak, 2021-04-09:
The problems tab in vscode should be ignored by end users show us the terminal tab the real errors are in red and prevent compile.

descipher, 2021-04-09:
is SDSUPPORT defined?

#if ENABLED(SDSUPPORT)
  int16_t DGUSScreenHandler::top_file = 0,
          DGUSScreenHandler::file_to_print = 0;
  static ExtUI::FileList filelist;
#endif

Vertabreak, 2021-04-09:
i used the provided example direct from GitHub and it was enabled.
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-5%20Plus

descipher, 2021-04-09:
I have not synced for 24 hours but that's what it should have. I'll resync now to be sure.

Vertabreak, 2021-04-09:
here is a copy i'm working with just pulled bugfix and matching config for the ender 5 +
[Marlin-bugfix-2.0.x-ender5+.zip](https://github.com/MarlinFirmware/Marlin/files/6288621/Marlin-bugfix-2.0.x-ender5%2B.zip)

descipher, 2021-04-09:
https://github.com/MarlinFirmware/Marlin/pull/21485/files
Looks like its here, can't verify right now...

TheAZtech, 2021-04-10:
@Vertabreak, Thank you for that, i now know what to include and this is the same error messages I am getting. I learn quickly and now know for next time what to include. You'll are rockstars and I will be paying attention to this post for further information. let me know if anyone needs anything else.

descipher, 2021-04-11:
Definitely broken shortly after #21485 issues start showing up on #21500 for AVR boards.
This impacts:
DGUS_LCD_UI_ORIGIN, DGUS_LCD_UI_FYSETC, DGUS_LCD_UI_HIPRECY and DGUS_LCD_UI_MKS still has other issues compiling at the moment.
see:
Marlin\src\lcd\extui\lib\dgus\mks\DGUSScreenHandler.cpp: In static member function 'static void GcodeSuite::M1002()':
Marlin\src\lcd\extui\lib\dgus\mks\DGUSScreenHandler.cpp:1274:22: error: no matching function for call to 'GcodeSuite::set_e_relative(bool)'
   set_e_relative(true); // M83
                      ^

D337z, 2021-04-18:
And here's something that seems to be related, but I'm not sure.
Might looks at changing over to U8G2

`Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\lcd\dogm/marlinui_DOGM.h:31:0,

                 from sketch\src\lcd\marlinui.cpp:141:

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:37:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:40:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7565_64128n_HAL_2x_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:53:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE

             ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_4X_HAL::init(pin_t, pin_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:56:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_HAL_4x_hw_spi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset); // a0 = U8G_PIN_NONE

             ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_ST7920_128X64_RRD::init(pin_t, pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:71:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_st7920_128x64_rrd_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset);   // a0 = U8G_PIN_NONE

             ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_SH1106_128X64_2X_I2C_2_WIRE::init(uint8_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:81:59: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_sh1106_128x64_2x_i2c_2_wire, options); }

                                                           ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1306_128X64_2X_I2C_2_WIRE::init(uint8_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:90:59: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t options = U8G_I2C_OPT_NONE) { U8GLIB::init(&u8g_dev_ssd1306_128x64_2x_i2c_2_wire, options); }

                                                           ^~~~

sketch\src\lcd\dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_TFT_320X240_UPSCALE_FROM_128X64::init(uint8_t, uint8_t, uint8_t)':

sketch\src\lcd\dogm/HAL_LCD_class_defines.h:103:77: error: 'init' is not a member of 'U8GLIB'

   void init(uint8_t cs, uint8_t rs, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_tft_320x240_upscale_from_128x64, cs, rs, reset); }
[... 320 lines cut ...]

sketch\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm\HAL_LCD_class_defines.h:131:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_ssd1309_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

sketch\src\lcd\dogm\HAL_LCD_class_defines.h: In member function 'void U8GLIB_SSD1309_128X64_HAL::init(pin_t, pin_t, pin_t)':

sketch\src\lcd\dogm\HAL_LCD_class_defines.h:134:13: error: 'init' is not a member of 'U8GLIB'

     U8GLIB::init(&u8g_dev_ssd1309_hw_spi, (uint8_t)cs, (uint8_t)a0, (uint8_t)reset);

             ^~~~

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

ellensp, 2021-05-16:
@D337z you cannot use the U8G lib that arduino uses, you have to to the marlin fork of U8G https://github.com/MarlinFirmware/U8glib-HAL 
Easiest way to do that is to use platformio, then it automatically uses the correct library. 
This is also unrelated as the Ender 5 Plus does not use that type of LCD

ellensp, 2021-05-16:
Can Someone try this diff, it compiles, but I dont have the lcd.
Check what happens whey your printing from sd and pause printing from lcd

```
diff --git a/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp b/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
index 1c2944bb4f..31bcaac236 100644
--- a/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
+++ b/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
@@ -194,9 +194,8 @@ const struct DGUS_VP_Variable ListOfVP[] PROGMEM = {

   // Temperature Data
   #if HOTENDS >= 1
-    VPHELPER(VP_T_E0_Is, nullptr, nullptr, SET_VARIABLE(getActualTemp_celsius, E0, long)),
-    VPHELPER(VP_T_E0_Set, nullptr, GET_VARIABLE(setTargetTemp_celsius, E0),
-                                   SET_VARIABLE(getTargetTemp_celsius, E0)),
+    VPHELPER(VP_T_E0_Is, &thermalManager.temp_hotend[0].celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>),
+    VPHELPER(VP_T_E0_Set, &thermalManager.temp_hotend[0].target, ScreenHandler.HandleTemperatureChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
     VPHELPER(VP_Flowrate_E0, nullptr, ScreenHandler.HandleFlowRateChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
     VPHELPER(VP_EPos, &destination.e, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<2>),
     VPHELPER(VP_MOVE_E0, nullptr, &ScreenHandler.HandleManualExtrude, nullptr),
diff --git a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
index f05dfc6f70..fe7c4120b8 100644
--- a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
+++ b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
@@ -85,7 +85,7 @@

       case 1: // Pause

-        GotoScreen(MKSLCD_SCREEN_PAUSE);
+        GotoScreen(DGUSLCD_SCREEN_POPUP);
         if (!ExtUI::isPrintingFromMediaPaused()) {
           ExtUI::pausePrint();
           //ExtUI::mks_pausePrint();
```

nicwilson58, 2021-05-19:
> ```diff
> VPHELPER(VP_T_E0_Is, &thermalManager.temp_hotend[0].celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>),
> +    VPHELPER(VP_T_E0_Set, &thermalManager.temp_hotend[0].target, ScreenHandler.HandleTemperatureChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
>      
> ```

I compiled this for the CR10s Pro V2 which had the same compile errors as the Ender5+ had using this DGUS version,  and with your changes above it compiles and works for the LCD,  however when setting the temperatures,   the nozzle shows correct,  but bed heats but shows random temperature values on the LCD, which given the changes shown looks like a possible issue with the changes.

ellensp, 2021-05-19:
@nicwilson58 I don't understand you comments. The edits where for setting E0 temp, nothing to do with the bed...

Only thing I can see is perhaps the float is causing issues...  
E1 and the bed both use DGUSLCD_SendWordValueToDisplay
where as E0 used DGUSLCD_SendFloatAsLongValueToDisplay
Also spotted an issue with HOTENDS >= 2 and corrected this.

Try this version

```DIFF
--- a/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
+++ b/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
@@ -194,9 +194,8 @@ const struct DGUS_VP_Variable ListOfVP[] PROGMEM = {
 
   // Temperature Data
   #if HOTENDS >= 1
-    VPHELPER(VP_T_E0_Is, nullptr, nullptr, SET_VARIABLE(getActualTemp_celsius, E0, long)),
-    VPHELPER(VP_T_E0_Set, nullptr, GET_VARIABLE(setTargetTemp_celsius, E0),
-                                   SET_VARIABLE(getTargetTemp_celsius, E0)),
+    VPHELPER(VP_T_E0_Is, &thermalManager.temp_hotend[0].celsius, nullptr, ScreenHandler.DGUSLCD_SendWordValueToDisplay),
+    VPHELPER(VP_T_E0_Set, &thermalManager.temp_hotend[0].target, ScreenHandler.HandleTemperatureChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
     VPHELPER(VP_Flowrate_E0, nullptr, ScreenHandler.HandleFlowRateChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
     VPHELPER(VP_EPos, &destination.e, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<2>),
     VPHELPER(VP_MOVE_E0, nullptr, &ScreenHandler.HandleManualExtrude, nullptr),
@@ -216,7 +215,7 @@ const struct DGUS_VP_Variable ListOfVP[] PROGMEM = {
     #endif
   #endif
   #if HOTENDS >= 2
-    VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, DGUSLCD_SendWordValueToDisplay),
+    VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, ScreenHandler.DGUSLCD_SendWordValueToDisplay),
diff --git a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
index f05dfc6f70..fe7c4120b8 100644
--- a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
+++ b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
@@ -85,7 +85,7 @@

       case 1: // Pause

-        GotoScreen(MKSLCD_SCREEN_PAUSE);
+        GotoScreen(DGUSLCD_SCREEN_POPUP);
         if (!ExtUI::isPrintingFromMediaPaused()) {
           ExtUI::pausePrint();
           //ExtUI::mks_pausePrint();
 ```

borjarodi, 2021-05-19:
> @nicwilson58 I don't understand your comment. The edits where for setting E0 temp, nothing to do with the bed...
> 
> Only thing I can see is perhaps the float is causing issues...
> E1 and the bed both use DGUSLCD_SendWordValueToDisplay
> where as E0 used DGUSLCD_SendFloatAsLongValueToDisplay
> Also spotted an issue with HOTENDS >= 2 and corrected this.
> 
> Try this version
> 
> ```diff
> --- a/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
> +++ b/Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp
> @@ -194,9 +194,8 @@ const struct DGUS_VP_Variable ListOfVP[] PROGMEM = {
>  
>    // Temperature Data
>    #if HOTENDS >= 1
> -    VPHELPER(VP_T_E0_Is, nullptr, nullptr, SET_VARIABLE(getActualTemp_celsius, E0, long)),
> -    VPHELPER(VP_T_E0_Set, nullptr, GET_VARIABLE(setTargetTemp_celsius, E0),
> -                                   SET_VARIABLE(getTargetTemp_celsius, E0)),
> +    VPHELPER(VP_T_E0_Is, &thermalManager.temp_hotend[0].celsius, nullptr, ScreenHandler.DGUSLCD_SendWordValueToDisplay),
> +    VPHELPER(VP_T_E0_Set, &thermalManager.temp_hotend[0].target, ScreenHandler.HandleTemperatureChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
>      VPHELPER(VP_Flowrate_E0, nullptr, ScreenHandler.HandleFlowRateChanged, &ScreenHandler.DGUSLCD_SendWordValueToDisplay),
>      VPHELPER(VP_EPos, &destination.e, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<2>),
>      VPHELPER(VP_MOVE_E0, nullptr, &ScreenHandler.HandleManualExtrude, nullptr),
> @@ -216,7 +215,7 @@ const struct DGUS_VP_Variable ListOfVP[] PROGMEM = {
>      #endif
>    #endif
>    #if HOTENDS >= 2
> -    VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, DGUSLCD_SendWordValueToDisplay),
> +    VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, ScreenHandler.DGUSLCD_SendWordValueToDisplay),
> diff --git a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
> index f05dfc6f70..fe7c4120b8 100644
> --- a/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
> +++ b/Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp
> @@ -85,7 +85,7 @@
> 
>        case 1: // Pause
> 
> -        GotoScreen(MKSLCD_SCREEN_PAUSE);
> +        GotoScreen(DGUSLCD_SCREEN_POPUP);
>          if (!ExtUI::isPrintingFromMediaPaused()) {
>            ExtUI::pausePrint();
>            //ExtUI::mks_pausePrint();
> ```

In this code when the rows start with "-" they have to be deleted and when they start with "+" they have to be added? I am a beginner, thanks

borjarodi, 2021-05-19:
Thanks to the comments, I have managed to compile the code, but with a warning, which I have attached in an image. Now the problem is that when I turn on the Ender 5 Plus the screen is frozen. Any solution? Thank you very much for your help. 
[image]
[image]

nicwilson58, 2021-05-21:
> @nicwilson58 I don't understand your comment. The edits where for setting E0 temp, nothing to do with the bed...
I see that now,  I think the confusion for me came from the way Marlin has always been that I have seen,  if you go to PID tuning for the hotend its always been shown as E1 not E0
[image]

ellensp, 2021-05-21:
yes boards are e0, e1 etc and on LCD are E1, E2 etc,  this has always irked me also...

nicwilson58, 2021-05-21:
> yes boards are e0, e1 etc and on LCD are E1, E2 etc, this has always irked me also...

Thanks,  the changes seem to work,  next to find out why the tools menu does nothing.

D337z, 2021-05-21:
The bed, if set is E-1. And I don't have a standard Ender 5 Plus. It's
heavily modified.

nicwilson58, 2021-05-21:
> Thanks to the comments, I have managed to compile the code, but with a warning, which I have attached in an image. Now the problem is that when I turn on the Ender 5 Plus the screen is frozen. Any solution? Thank you very much for your help.
> 
> [image]
> 
> [image]

try using the matching screen firmware.

ellensp, 2021-05-22:
@nicwilson58 the question remains should I turn my patch into a pr, is some functionality is better than none at all...

shaders, 2021-06-22:
`VPHELPER(VP_T_E0_Is, &thermalManager.temp_hotend[0].celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>),`

and
`VPHELPER(VP_T_Bed_Is, &thermalManager.temp_bed.celsius, nullptr, ScreenHandler.DGUSLCD_SendFloatAsLongValueToDisplay<0>),`

Is actually correct code. Otherwise display shows garbage numbers.

CRCinAU, 2021-07-10:
Checking the latest bugfix-2.0.x branch for both configs and Marlin code today, and I get the following errors:

```
Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp:277:1: error: no matches converting function 'GetVariable' to type 'void (*)(struct DGUS_VP_Variable&, void*)'
 };
 ^
In file included from Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp:32:0:
Marlin/src/lcd/extui/dgus/origin/../DGUSDisplay.h:76:15: note: candidate is: template<class T, void (* Setter)(float, T), T selector> static void DGUSDisplay::GetVariable(DGUS_VP_Variable&, void*)
   static void GetVariable(DGUS_VP_Variable &var, void *val_ptr) {
               ^~~~~~~~~~~
Compiling .pio/build/mega2560/src/src/module/probe.cpp.o
*** [.pio/build/mega2560/src/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp.o] Error 1
Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_ResumePauseAbort(DGUS_VP_Variable&, void*)':
Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp:88:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope
         GotoScreen(MKSLCD_SCREEN_PAUSE);
                    ^~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp:88:20: note: suggested alternative: 'DGUSLCD_SCREEN_UNUSED'
         GotoScreen(MKSLCD_SCREEN_PAUSE);
                    ^~~~~~~~~~~~~~~~~~~
                    DGUSLCD_SCREEN_UNUSED
*** [.pio/build/mega2560/src/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp.o] Error 1
```

I didn't want to start another Issue, as this may well be related - but its reproduced by using both the stock config and todays bugfix-2.0.x checkout.

ellensp, 2021-07-11:
GetVariable and SetVariable where added in https://github.com/MarlinFirmware/Marlin/pull/18718
a proposal that was never actually finished, that got merged anyway... :( 
Reverting that part of the code and you get to the last error MKSLCD_SCREEN_PAUSE which is simply not defined for this dispaly.
It should be DGUS_SCREEN_PAUSE, but that is also not defined. I don't know how you would find out what screen number this should be set to...  so it sits waiting.

CRCinAU, 2021-07-11:
It doesn't look like the original PR author has been active since August 2020 - as such, pinging @thinkyhead to see how we should get to the bottom of this...

Hookedon3D, 2021-07-24:
Any update on this? I am trying to compile for my stock E5+ and I am getting the same error. Thanks in advance!

ellensp, 2021-07-30:
I have created a PR to get basic functionally back working on Marlin bugfix.

I had to purchase this LCD especially for this and as such I am asking if this helps you out to consider tipping me for my time an efforts and to recover some of the parts cost.  https://www.buymeacoffee.com//marlin

Hookedon3D, 2021-07-30:
I just redownloaded everything and tried still the same error "Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:277:1: error: no matches converting function 'GetVariable' to type 'void (*)(struct DGUS_VP_Variable&, void*)'"

ellensp, 2021-07-31:
@Hookedon3D This was only merged 5 minutes ago, so is only just now been added to bugfix. 
A PR is a pull request, code waiting review by the admins before being added (merged) into Marlin.  
You need to manually apply PR's to your code to try them, or replicate the code changes you want. Or wait for them to be merged.

closed 2021-07-31 by ellensp

Hookedon3D, 2021-07-31:
Oh ok, looks like it's merged? So, I'll go ahead and redownload it and try it again. Thank you!