MarlinFirmware/Marlin #21580: [FR] When you will add BTT E3 RRF V1.1 into bugfix Marlin 2.0.x?

trengtor, 2021-04-10:
### Describe the feature you want

Hi, guys! 
When you will add BTT E3 RRF V1.1 into bugfix Marlin 2.0.x?

thisiskeithb, 2021-04-10:
@bigtreetech?

trengtor, 2021-04-10:
> @bigtreetech?

https://www.biqu.equipment/collections/all-product/products/bigtreetech-e3-rrf-v1-1-mainboard-3d-printer-part-integrated-esp8266-wi-fi-module-for-ender3?variant=39251709427810

thisiskeithb, 2021-04-10:
> Interested in support here. Not by a fork created in BTT's github.

I'm not sure what you mean by that. I tagged BigTreeTech to ask when they plan to submit a PR to add support.

I'm busy with their other boards & printers right now (among many other projects) or I'd have a PR in by now.

peppekerstens, 2021-04-11:
Same request here.

[I tried to compare the pinout of the BTT with the SKR-PRO board]( https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/Schematic/SKR-PRO-V1.2.PDF), which has the same chip. But at this moment, there seems to be very little info for this new board, [except what's available on their site](https://www.biqu.equipment/products/bigtreetech-e3-rrf-v1-1-mainboard-3d-printer-part-integrated-esp8266-wi-fi-module-for-ender3?variant=39251709427810)

Anyways, I tried to make it work with the SKR-PRO board definition within Marlin, but no dice :(

You may check provided files as reference (again; does NOT work!)
I may have made a mistake here or the as I am no Marlin wizard. The configuration is based upon my own config for Ender5-SKR1.3-TMC2208-BLTOUCH combo (working).
[BigTreeTech E3 RRF 1.1 TMC22209.zip](https://github.com/MarlinFirmware/Marlin/files/6292053/BigTreeTech.E3.RRF.1.1.TMC22209.zip)

trengtor, 2021-04-13:
> Same request here.
> 
> [I tried to compare the pinout of the BTT with the SKR-PRO board](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/Schematic/SKR-PRO-V1.2.PDF), which has the same chip. But at this moment, there seems to be very little info for this new board, [except what's available on their site](https://www.biqu.equipment/products/bigtreetech-e3-rrf-v1-1-mainboard-3d-printer-part-integrated-esp8266-wi-fi-module-for-ender3?variant=39251709427810)
> 
> Anyways, I tried to make it work with the SKR-PRO board definition within Marlin as it made sense to me that BTT would not re-design the whole pinout, but no dice :(
> 
> You may check provided files as reference (again; does NOT work!)
> I may have made a mistake here or the as I am no Marlin wizard. The configuration is based upon my own config for Ender5-SKR1.3-TMC2208-BLTOUCH combo (working), which in turn is based on the Creality/Ender-5/BigTreeTech SKR Mini E3 2.0 example.
> [BigTreeTech E3 RRF 1.1 TMC22209.zip](https://github.com/MarlinFirmware/Marlin/files/6292081/BigTreeTech.E3.RRF.1.1.TMC22209.zip)
> 
> During build, many warnings on missing example reference, but no errors. Partial buildlog in zip

BTT alllready made own fork for it. But have wantings to use single Marlin repo instead many forks.

ondrada, 2021-04-14:
[e3-rrf.zip](https://github.com/MarlinFirmware/Marlin/files/6314005/e3-rrf.zip)
Hello,

in attached ZIP are some new and changed files against current 2.0.x bugfix. They are trying to bring support for BTT E3 RRF and are based on BigTreeTech E3 RRF fork of Marlin plus their and CR10 SKR Pro 1.2 config files. 

It should be enough to get current 2.0.x bugfix, add/overwrite what is in attached ZIP and compile with platformio:
`platformio run -e BIGTREE_E3_RRF`

Its definitely compilable, but I've no clue if it really works, as my E3 RRF board is still in hands of our "superfast" czech postal service :-) So good luck :-)

closed 2021-04-16 by thinkyhead