MarlinFirmware/Marlin #21581: Creality 4.2.7 board PIN out

jarjartee, 2021-04-10:
### Is your feature request related to a problem? Please describe.

I know this is not a Marlin problem but grappling in the dark here(also contacted Creality). Trying to set hot end fan to go on at 50°C but can NOT find correct PIN out - anyone got the diagrams/pin out list ????

### Are you looking for hardware support?

Creality 4.2.7 board PIN out diagram

### Describe the feature you want

How to control the HOTEND fan

ellensp, 2021-04-10:
That board only has 1 software controllable fan PA0  this is used for parts cooling. 
The K-fans are on when power is on. 
You would have to add a external mosfet and connect it somewhere to do this on this controller

closed 2021-04-10 by ellensp

jarjartee, 2021-04-10:
thank you very much for your assistance