MarlinFirmware/Marlin #21582: [BUG] Maximum temperature not enforced in thermal manage_heaters and PID autotune

Sebazzz, 2021-04-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The maxtemp values are not initialized:

[image]

This causes thermal runaway not to function correctly in `thermalManager::manage_heaters` and also cause PID auto-tune not to enforce.

I've modified the code in pid autotune to dump the maximum temperature, and it shows:

    Max temp:16383

```C++

    SERIAL_ECHOLNPAIR("Max temp:",temp_range[heater_id].maxtemp);
    if (target > GHV(CHAMBER_MAX_TARGET, BED_MAX_TARGET, temp_range[heater_id].maxtemp - (HOTEND_OVERSHOOT))) {
      SERIAL_ECHOLNPGM(STR_PID_TEMP_TOO_HIGH);
      TERN_(EXTENSIBLE_UI, ExtUI::onPidTuning(ExtUI::result_t::PID_TEMP_TOO_HIGH));
      return;
    }

    SERIAL_ECHOLNPGM(STR_PID_AUTOTUNE_START);

    return;

```

### Bug Timeline

Appears to have started at b35bfeb1c3f1271016227a1d79dcb552e67e3cb9 for #21016

### Expected behavior

PID autotune should show when given a too high temperature:

    PID Autotune failed! Temperature too high

### Actual behavior

Instead is shown:

    PID Autotune start

And the hotend is heated up far before HEATER_0_MAXTEMP (275 in my case).

### Steps to Reproduce

1. Use the default configuration supplied, or any of the sample configurations from the Configurations repository. It doesn't matter.
2. Run PID auto-tune:

    M303 S900

3. Note the hotend heats up beyond the maximum allowed value set in the firmware

### Version of Marlin Firmware

Latest

### Printer model

Creality CR-6 SE

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Sebazzz, 2021-04-10:
Changing 

    #define _MINMAX_TEST(N,M) (HOTENDS > N && TEMP_SENSOR_ ##N## THERMISTOR_ID && TEMP_SENSOR_ ##N## THERMISTOR_ID != 998 && TEMP_SENSOR_ ##N## THERMISTOR_ID != 999 && defined(HEATER_##N##_##M##TEMP))

To

    #define _MINMAX_TEST(N,M) (HOTENDS > N && TEMP_SENSOR_ ##N## _THERMISTOR_ID && TEMP_SENSOR_ ##N## THERMISTOR_ID != 998 && TEMP_SENSOR_ ##N## THERMISTOR_ID != 999 && defined(HEATER_##N##_##M##TEMP))

Note the `_` that allows to resolve `TEMP_SENSOR_0_THERMISTOR_ID` instead of `TEMP_SENSOR_0THERMISTOR_ID` fixes the issue.

However, the `_TEMP_MIN_E` macro fails:

```
Compiling .pio\build\STM32F103RE_btt_USB\src\src\module\temperature.cpp.o
In file included from Marlin\src\module\../inc/../core/boards.h:24:0,
                 from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\temperature.cpp:30:
Marlin\src\module\temperature.cpp: In member function 'void Temperature::init()':
Marlin\src\module\temperature.cpp:2211:127: error: 'TEMP_SENSOR_0_MINTEMP_IND' was not declared in this scope
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                                                                                               ^
Marlin\src\module\../inc/../core/macros.h:514:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION ? 'A' : 'B'
                             ^~~~~
Marlin\src\module\temperature.cpp:2211:58: note: in expansion of macro 'TERN'
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                          ^~~~
Marlin\src\module\temperature.cpp:2211:96: note: in expansion of macro 'pgm_read_word'
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                                                                ^~~~~~~~~~~~~
Marlin\src\module\temperature.cpp:2226:8: note: in expansion of macro '_TEMP_MIN_E'
        _TEMP_MIN_E(0);
        ^~~~~~~~~~~
Marlin\src\module\temperature.cpp:2211:127: note: suggested alternative: 'TEMP_SENSOR_0_MAXTEMP_IND'
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                                                                                               ^
Marlin\src\module\../inc/../core/macros.h:514:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION ? 'A' : 'B'
                             ^~~~~
Marlin\src\module\temperature.cpp:2211:58: note: in expansion of macro 'TERN'
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                          ^~~~
Marlin\src\module\temperature.cpp:2211:96: note: in expansion of macro 'pgm_read_word'
       const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius))); \
                                                                                                ^~~~~~~~~~~~~
Marlin\src\module\temperature.cpp:2226:8: note: in expansion of macro '_TEMP_MIN_E'
        _TEMP_MIN_E(0);
        ^~~~~~~~~~~
*** [.pio\build\STM32F103RE_btt_USB\src\src\module\temperature.cpp.o] Error 1
```

ellensp, 2021-04-11:
Marlin/src/module/thermistor/thermistors.h is quite broken..

looks like 
```CPP
#ifdef TEMPTABLE_0
  #if TT_REV(0)
    #define TEMP_SENSOR_0_MINTEMP_IND 0
    #define TEMPTABLE_0_MAXTEMP_IND HEATER_0_LEN - 1
  #else
    #define TEMPTABLE_0_MINTEMP_IND HEATER_0_LEN - 1
    #define TEMP_SENSOR_0_MAXTEMP_IND 0
  #endif
#endif
```
Should be 
```CPP
#ifdef TEMPTABLE_0
  #if TT_REV(0)
    #define TEMP_SENSOR_0_MINTEMP_IND 0
    #define TEMP_SENSOR_0_MAXTEMP_IND TEMPTABLE_0_LEN - 1
  #else
    #define TEMP_SENSOR_0_MINTEMP_IND TEMPTABLE_0_LEN - 1
    #define TEMP_SENSOR_0_MAXTEMP_IND 0
  #endif
#endif
```

Same with TEMPTABLE_0  through TEMPTABLE_7

Still getting some warnings I dont like...  but have to stop for today.
Perhaps this will give someone a clue to finish it.

ellensp, 2021-04-11:
Please test PR https://github.com/MarlinFirmware/Marlin/pull/21592

Sebazzz, 2021-04-11:
Will do. I also feel like this could use some static assert or something to prevent any mistake from allowing to happen in the future 😊

Sebazzz, 2021-04-11:
@ellensp Good work, the fix works!

ellensp, 2021-04-13:
My PR has been merged. This should be sorted.

closed 2021-04-13 by ellensp