MarlinFirmware/Marlin #21583: [BUG] Incorrect Z position after Auto Bed Leveling (HOME right)

Gabrielerusso, 2021-04-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin applies an incorrect Z offset after bed leveling, when using the simple "home" the Z axis offset is perfect, i printed like  that for a month, but then i tried the auto bed leveling feature and it ruined a new PEI sticker...
The print bed, after homing, it's off by 0.85mm all the times, the right Z offset, that works with home, is -1.85mm but after the auto bed leveling the bed it's 0.85mm higher than it should be, so i need to set the offset at -1mm after bed leveling and -1.85mm when bed leveling is disabled, this is a nosense.
I tested it with Marlin 2.0.7.2 and also with bugfix-2.0.x, doing a quick research i found out that i wasn't the online one in those years to get the same problem, and i didn't find any solution.

### Bug Timeline

From a google research this is a bug or problem present since 2016

### Expected behavior

I expected to get a leveled bed and good adesion, not a nozzle scratching the PEI bed, actually if i put the nozzle at the right height it works pretty well

### Actual behavior

nozzle scratching the PEI bed, as it's 0.85mm lower than it should be

### Steps to Reproduce

-Home Z 
-Move Z to 0
-put a paper between nozzle and bed to check homing, and it's perfect
-Auto bed leveling 
-Move Z to 0.8 and the nozzle it's already touching the bed
-Move Z to 0 and the nozzle it's pushing the bed down of around 0.8mm

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

custom prusa bear like frame

### Electronics

mks sgenL V2

### Add-ons

TriangleLab 3D-Touch

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

thisiskeithb, 2021-04-10:
Please attach your configs.

---

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Gabrielerusso, 2021-04-10:
i was doing all sort of test with different configurations to provide you more information possible when i noticed there was "#define Z_MIN_POS -1" still configured from the previous configuration. I replaced this value with something like -10 just to have a pretty clear visual reference and the problem was this!
Z_MIN_POS was not affecting the 3d printer when Homing, but was being used after auto bed leveling. If i'm right this is not documented anywhere and to me looks like an unwanted feature.
I don't actually have the time to look into the code itself, but if you understood what happened or if you wan't some specific informations please tell me and i will try to give them.

> Please attach your configs.
>
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

wwoody777, 2021-04-13:
I have exactly the same problem, which I could observe several times yesterday, after upgrading the firmware 2 days ago. In my configuration Z_MIN_POS is 0 (default) and after PROBE WIZZARD (bltouch z offset measured with caliper exactly -2.88) and BED LEVELING, I have to increase the height of the Z axis by exactly 1 mm to -1.88 mm, otherwise the nozzle drills into the bed.
[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/6302206/Marlin_Config.zip)

wwoody777, 2021-04-13:
Just tested. Same configuration files but Marlin-bugfix-2.0.x from March 29, 2021. EEPROM initialized, everything reset, auto Z-align performed, corners leveled, probe wizzard performed, bed leveled and it works like a dream.

Gabrielerusso, 2021-04-13:
> Just tested. Same configuration files but Marlin-bugfix-2.0.x from March 29, 2021. EEPROM initialized, everything reset, auto Z-align performed, corners leveled, probe wizzard performed, bed leveled and it works like a dream.

Maybe there was an offset set in EEPROM and after the re initialization it got back to default values.

wwoody007, 2021-04-13:
I initialize the EEPROM every time I flash a firmware and also do "reset to default" aka M502 and M500. Have looked everywhere in my configs to see if somehow, somewhere a 1 mm setting is set. Nope, nothing there.

JWest1969, 2021-04-26:
I have a gigabot 2.0 from RE:3D. I recently upgraded my printer to have a BTT SKR Pro V1.2 board with TMC2209 drivers, a BTT 35 E3 V3 display and a genuine BLTouch for bed probing. After running the G29 mesh probing routine used by ABL I notice the values stored seem incorrect. I know my bed is level and that there is a section that dips down and back up in one area. I expected to see this reflected in the mesh values but did not. I must use the mesh editor to touch off each point manually to get it working as expected. Before leveling manually, a test print is too close on the left side and too far away on the right. (along the X axis) I am curious as to why this is happening. I have seen several posts from people seeming to have similar problem but no solution. What I wonder is:
1. Are the z values stored in the list accurate?
2. If the values are accurate could the value itself being stored in the wrong index for the mesh?

I notice that when using ABL the G29 routine starts probing at index bottom right corner I0 J9 (10x10 mesh) and works its way back to I0 J0 then forward I1 J0 to I1 J9 in a zig zag pattern until it ends up at I9 J9. (Top right corner). I wonder if this is the intended pattern?  I would have thought the pattern should start at I0 J0 (bottom left) and end up at I9 J9 (top right) but I have no idea the intent. Could the program loop that controls the pattern position be incorrect and is scrambling up the stored values in the wrong actual position on the bed? Any help is greatly appreciated. It takes a lot of time to probe 100 points manually and is why I got the BL touch to automate the process.

tombrazier, 2022-02-23:
I suspect this issue is about not setting `NOZZLE_TO_PROBE_OFFSET` correctly and can be closed. The OP mentions solving the problem by setting "z_probe_offset".

I would recommend anyone having the same problem look at `NOZZLE_TO_PROBE_OFFSET` and also enable `PROBE_OFFSET_WIZARD` in `Configuration_adv.h` which makes it easy to determine the Z value for `NOZZLE_TO_PROBE_OFFSET`. Just make sure you have the latest bugfix if also using probe temperature compensation - otherwise the Z value will not take temp compensation into account.

closed 2022-05-06 by github-actions[bot]