MarlinFirmware/Marlin #21587: [BUG] (bug summary) Wrong Direction When Homing

agoy507, 2021-04-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

X axis Wrong direction when Homing other axis Normal, if homing x axis only, moving was normal. 
when jog all axis normal. whats wrong? I using esp32 board.

### Version of Marlin Firmware

2.0.7.2

### Printer model

Creality Ender 3

### Electronics

Esp32 Board

ellensp, 2021-04-11:
Please attach your configuration files.

agoy507, 2021-04-11:
That's my config file.
[Uploading Marlin-2.0.xOriEsp.zip…]()

msmeeh, 2021-04-26:
in configfile : just inver the direction of the x axis, if it is true change it to false, if it false change it to true 

#define INVERT_X_DIR false   // true

closed 2021-05-02 by thinkyhead